ev3dev/ev3dev #1539: Sensor driver for Building block robotics driver

Eisverygoodletter, 2022-08-15:
**Is your feature request related to a problem? Please describe.**
There is a new sensor from buildingblockrobotics.com [IrSeeker](https://irseeker.buildingblockrobotics.com/). It currently does not have an ev3dev driver

**Describe the solution you'd like**
An ev3dev driver to be added for it.

**Describe alternatives you've considered**
Point me to resources that can help me write the driver (i'm inexperienced in writing drivers)

**Additional context**
If the driver is written, will it be merged into ev3dev-stretch or just ev3dev-buster?

Eisverygoodletter, 2022-08-15:
Can definitions be simply added into https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-buster/sensors/nxt_i2c_sensor_defs.c or is there something more complex that needs to be done?

dlech, 2022-08-15:
If the new device follows the same patterns as other sensors, like having the vendor and device id in the first registers, then just adding a new device to the sensor defs should work (there are also some header files that need to be modified [example](https://github.com/ev3dev/lego-linux-drivers/commit/f935599393c5a5f6934ae957bc8a22ed04b7ced1)).

However, then you would have to compile your own kernel to be able to use it. It is probably easier to just use the `/dev/i2c*` kernel interface to communicate with the device.

Eisverygoodletter, 2022-08-16:
Thanks for the fast reply, how would I use the `/dev/i2c*` interface? Can
you recommend a guide or some documentation?

dlech, 2022-08-16:
There is some ev3dev-specific documentation at https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.htm

Eisverygoodletter, 2022-08-18:
https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.htm It says page not found

dlech, 2022-08-18:
https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.html