ev3dev/ev3dev #1541: Trying to upload my code from Visual Studio Code to my EV3

wessam33, 2022-08-22:
I'm trying to upload the demo code which LEGO provides in the "Getting started with EV3 MicroPython" document.
But every time I'm trying to send workspace to device the error message "Error sending file: Failure" pops up and won't allow me to download. I've tried looking online but there seems to be nothing there.

dlech, 2022-08-22:
What operating system? What connection method (USB/Bluetooth/WiFi)?

wessam33, 2022-08-22:
I'm using windows 11 and my connection method is Mini-USB.

WizGeek, 2022-08-26:
Things to check:

1. Is your MicroSD Card image "ev3dev-stretch-ev3-generic-2020-04-10" or "ev3-micropython-v2.0.0-sd-card-image"?
2. Are you using Visual Studio Code (VSCode)?
3. Did you install the VSCode extension "LEGO MINDSTORMS EV3 MicroPython"?  Did it also install "ev3dev-browser" v1.2.0?
4. Under the EV3DEV DEVICE BROWSER sidebar section, did you find and connect to your brick?
5. Are you using F5 or Run > Start Debugging?