ev3dev/ev3dev #1544: set default config for bluetooth adapter

Eisverygoodletter, 2022-09-24:
**Question**

When ev3dev starts up, if I run `hciconfig`, it will show me `UP RUNNNG PSCAN`. [`hci_config()`](https://linux.die.net/man/8/hciconfig) doesn't work with this for some reason, so I have to run `hciconfig hci0 piscan` so that it becomes `UP RUNNING PSCAN ISCAN` is enabled. This way, `hci_config()` works.
When I power off and then power on the ev3 brick, the setting goes back to `UP RUNNING PSCAN`, so I have to run the `hciconfig hci0 piscan` command again manually.
**Is there a way I can make `UP RUNNING PSCAN ISCAN` the default setting, or make the ev3 run a command (or bash script) on start up?**

**Background Information**
all commands are run via connecting to the robot using [putty](https://www.putty.org/).
I'm trying to use C to write a program on the ev3 brick which detects other ev3 bricks on bluetooth.

The code i'm using is
```c
// System 
#include <stdio.h>
#include <stdlib.h>

// Bluetooth
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <sys/socket.h>
#include <bluetooth/bluetooth.h>
#include <bluetooth/hci.h>
#include <bluetooth/hci_lib.h>

int
main() {

  // Get bluetooth device id
  int device_id = hci_get_route(NULL); // Passing NULL argument will retrieve the id of first avalaibe device 
  if (device_id < 0) {
    printf("Error: Bluetooth device not found");
    exit(1);
  }  
 
  // Find nearby devices
  int len     = 8;                 // Search time = 1.28 * len seconds
  int max_rsp = 255;               // Return maximum max_rsp devices
  int flags   = IREQ_CACHE_FLUSH;  // Flush out the cache of previously detected devices.
  inquiry_info* i_infs = (inquiry_info*) malloc(max_rsp * sizeof(inquiry_info)); 
  int num_rsp = hci_inquiry(device_id, len, max_rsp, NULL, &i_infs, flags); // search  
  if (num_rsp < 0) {
    printf("Error: the hci_inquiry fails");
    exit(1);
  }
  printf("Found %d device\n", num_rsp); 

  // Open socket
  int socket = hci_open_dev(device_id);
  if (socket < 0) {
    printf("Error: Cannot open socket");
    exit(1);
  }

  int i;
  for (i = 0; i < num_rsp; i++) {
    char device_address[20], device_name[300];
    inquiry_info* current_device = i_infs + i;
    // Get HEX address
    ba2str(&(current_device->bdaddr), device_address);
    // Clean previous name
    memset(device_name, 0, sizeof(device_name));
    // Get device name
    if (hci_read_remote_name(socket, &(current_device->bdaddr), sizeof(device_name), device_name, 0) < 0) {
        strcpy(device_name, "[unknown]"); // If cannot retrieve the name then set it "unknown"
    }
    // print address and name
    printf("%s  %s\n", device_address, device_name);
  }  

  // Close the socket
  close(socket);

  // Remove allocated memory
  free(i_infs);
  
  return 0;
}
```

dlech, 2022-09-24:
You can create a systemd service or use `/etc/rc.local` to run something at startup.

You should use the BlueZ D-Bus API or call the `bluetoothctl` command from a script instead of poking the HCI interface directly.

Eisverygoodletter, 2022-09-24:
Does poking the HCI interface directly cause safety concerns?

dlech, 2022-09-25:
It just goes behind the back of BlueZ (`bluetoothd`) so it confuse the user interface or anything else that uses BlueZ.

closed 2022-09-27 by Eisverygoodletter