ev3dev/ev3dev #1549: Brickman and OpenRoberta

BrunoESP32, 2022-12-10:
Hi,

I've been using the EV3dev on an EV3 robot for some time now and my experience has been very good. However, now I'm trying to run EV3dev on a Raspberry 2, but when I enabled OpenRoberta, the Brickman interface crashes. Apparently, the starting of the OpenRoberta service is delayed and this causes the problem (in fact, even if I plug in a keyboard, the interface is also crashes). The only way I found to get brickman.service back was to restart it using "systemclt". I wonder if there is a more elegant way to solve the problem instead of restarting brickman.service every time a new command line appears...

No Robot 
 - Device: only a Raspberry Pi 2
 - Versions: ev3dev-stretch-rpi2-generic-2020-04-10

dlech, 2022-12-12:
Are you using BrickPi or PiStorms? 

Can you debug the crash by installing the debug symbols packages for brickman and all of its dependencies and run it with `gdb`?

BrunoESP32, 2022-12-13:
Hi, David.

I'm not using BrickPi or PiStorms. I'm using a standard Raspberry Pi with a HDMI monitor, wifi dongle and a keyboard. 

I've been digging around trying to understand this, and i think Brickman is not crashing. When I enabladed the OpenRoberta, some service (maybe openrobertalab service) is writing on the main interface and that is disabling Brickman. This happens when openrobertalab is running, but i saw happening when USB host-controller driver (dwc_otg) writes something on the main interface too...