ev3dev/ev3dev #1551: I need help.

K1ZANA, 2023-01-12:
**What are you trying to do?**

I am trying to create a simple program with python, the control buttons and the touch sensor should be used to navigate it, the sensor should be able to select a testbox that just beeps when selected. (Using that for testing.)

**What did you expect to happen**

The program should have..started.

**What actually happened**

The program does not start.

**Source Code**

Here is the program I am trying to run.

```py
#!/usr/bin/env pybricks-micropython
from pybricks import ev3brick as brick
from pybricks.ev3devices import Motor, TouchSensor
from pybricks.parameters import Port, Stop
from pybricks.tools import wait
from random import randint

# Initialize variables
btn = TouchSensor(Port.S1)
lcd = brick.display()
motor = Motor(Port.A)

# Set cursor position and size
cursor_x = 0
cursor_y = 0
cursor_size = 1

# Set box position and size
box_x = 50
box_y = 50
box_size = 50

# Set box text
box_text = "TestBox"

# Set outline color
outline_color = "black"

while True:
    # Clear the display
    lcd.clear()

    # Draw the box with text in the middle
    lcd.draw.rectangle((box_x, box_y, box_x + box_size, box_y + box_size), outline=outline_color)
    lcd.draw.text((box_x + (box_size / 2), box_y + (box_size / 2)), box_text)

    # Draw the cursor
    lcd.draw.ellipse((cursor_x, cursor_y, cursor_x + cursor_size, cursor_y + cursor_size), outline=outline_color)

    # Update the display
    lcd.update()

    # Check if the cursor is hovering over the box
    if cursor_x > box_x and cursor_x < box_x + box_size and cursor_y > box_y and cursor_y < box_y + box_size:
        outline_color = "red"
        cursor_size = 2
    else:
        outline_color = "black"
        cursor_size = 1

    # Check for button presses
    if btn.pressed():
        motor.run_time(500, 1000, Stop.BRAKE, True)
        cursor_x += 5
    if btn.released():
        motor.run_time(-500, 1000, Stop.BRAKE, True)
        cursor_x -= 5

    # Check if cursor is over the box
    if cursor_x > box_x and cursor_x < box_x + box_size and cursor_y > box_y and cursor_y < box_y + box_size:
        # Move the cursor to the center of the box
        cursor_x = box_x + (box_size / 2)
        cursor_y = box_y + (box_size / 2)
        # Play a sound to indicate selection
        brick.sound.beep()
        # Move the box to a random location on the screen
        box_x = randint(0, lcd.width - box_size)
        box_y = randint(0, lcd.height - box_size)
    else:
        # Play a sound to indicate that cursor is not over the box
        brick.sound.beep()
        wait(100)
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: Windows 11 22H2
 - Connection Type: USB
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 

```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534C5833
BOARD_INFO_TYPE=main

```

**Background Information**

See `What are you trying to do?`.

**Additional Problems**

I am also not able to install any package, since the sources.list just doesnt work, i cant install pip, i cant install anything.

dlech, 2023-01-13:
The best place to get Pybricks support is https://github.com/orgs/pybricks/discussions. Can you copy the issue there and add information on what error messages you are seeing?

K1ZANA, 2023-01-13:
> To install stuff on ev3dev, it has to be connected to the internet. However, the Python version is too old, so pip is no longer compatible with PyPI. You can only install debian packages.

...yeah i think thats my error, but how do i connect it to my internet without bluetooth?

> The best place to get Pybricks support is https://github.com/orgs/pybricks/discussions. Can you copy the issue there and add information on what error messages you are seeing?

Sure, gonna do that right now.

dlech, 2023-01-13:
If you don't want to use Bluetooth, you can use USB or Wi-Fi to connect to the internet.

See https://www.ev3dev.org/docs/networking/