ev3dev/ev3dev #1552: Connecting Multiple Ev3dev

pjaswin, 2023-01-13:
**Question**

How to communicate 3 ev3dev devices each other?

**Background Information**

I am trying to demonstrate a vehicle communication system using the ev3dev system. But we are using 3 bricks to communicate with each other. we don't know how can we communicate with each other. Please give us suggestion if there is any method

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2023-01-13:
https://pybricks.com/ev3-micropython/messaging.html#making-bigger-networks

pjaswin, 2023-01-20:
> ###
> [Jan 13, 2023](#issuecomment-1382137000)

@dlech Is it possible to establish communication between two bricks using MQTT?