ev3dev/ev3dev #1553: NXT Color Sensor support?

TheDoctor2017, 2023-01-15:
ev3dev version: 4.14.117-ev3dev-2.3.5-ev3
ev3dev-lang-python version:
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name Version Architecture Description
ii micropython-ev3dev2 2.1.0 all Python language bindings for ev3dev for MicroPython
ii python3-ev3dev 1.2.0 all Python language bindings for ev3dev
ii python3-ev3dev2 2.1.0 all Python language bindings for ev3dev

I am trying to use an NXT color sensor, but I can't get the EV3 brick to recognize it. I know the sensor is good, because if i boot the EV3 into the default firmware from Lego the sensor is detected and work properly, but when I boot to EV3DEV and check Device Browser/Sensors the sensor does not show. Note that this is the NXT color sensor, 4546542, that comes with the NXT 2.0 kit, not the ambient light sensor. Is this sensor supported? Thanks!

dlech, 2023-01-15:
Can you share the code to reproduce the problem?

TheDoctor2017, 2023-01-16:
> Can you share the code to reproduce the problem?

When I try to initialize the NXT color sensor using MicroPython using:

color_sensor = ColorSensor(Port.S1)

I get the error:

A sensor or motor is not connected to the specified port

dlech, 2023-01-16:
> When I connect the NXT color sensor and boot using EV3DEV, then go into Device Browser/Sensors, the sensor does not show as connected

It cannot be automatically detected and there is no kernel driver for it, so this sensor will never display in the device browser. The Pybricks library takes care of properly configuring the port when you run your program.

> using MicroPython, not Pybricks

Pybricks is built on top of MicroPython, so they are basically the same thing in this context.

> If not, where do I find Pybricks for the EV3?

It is pre-installed on ev3dev. Just use `#!/usr/bin/env pybricks-micropython` as the first line of your program to use the Pybricks MicroPython runtime.

> A sensor or motor is not connected to the specified port

Can you share the full program to reproduce the problem?

TheDoctor2017, 2023-01-16:
> Can you share the full program to reproduce the problem?

This is the entire code:

#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()

# Write your program here.
ev3.speaker.beep()

color_sensor = ColorSensor(Port.S1)

dlech, 2023-01-16:
This is the NXT color sensor, so you need `from pybricks.nxtdevices import ColorSensor`, not `ev3devices`.

TheDoctor2017, 2023-01-16:
> This is the NXT color sensor, so you need `from pybricks.nxtdevices import ColorSensor`, not `ev3devices`.

Wow. Sorry I missed that. Thank you so much!