ev3dev/ev3dev #1554: Will Debian Bullseye come?

jwaxy, 2023-01-28:
**Question**

Debian released Buster and Bullseye but ev3dev is stuck on stretch. Does there is any plan to upgrade? Or is this project discounted?

dlech, 2023-01-30:
Not likely (unless someone wants to sponsor me to work on it).

BigKitten, 2023-08-30:
I wanted to try to update it myself, any suggestions to how I might do it?

dlech, 2023-08-30:
The first step would be to get https://github.com/ev3dev/lego-linux-drivers updated to the latest kernel and make sure everything still works. Then try building the image from https://github.com/ev3dev/docker-library and see if there are any more ev3dev-specific packages that need to be updated and rebuilt.

BigKitten, 2023-08-30:
Thanks! I’ll try that!

Growflavor, 2023-12-09:
> Thanks! I’ll try that!

@pythoncat1 I am also curious about this...did you happen to make any progress with either Bullseye (or Buster)?

atk-r, 2023-12-10:
@Growflavor Unfortunately no… i’m not really familiar with this sort of thing 😅

pythoncat1, 2023-12-10:
@Growflavor Unfortunately no… i’m not really familiar with this sort of thing 😅

Growflavor, 2023-12-11:
> @Growflavor Unfortunately no… i’m not really familiar with this sort of thing 😅

Ok thank you for the update!  

Some reference info for anyone else curious or tinkering:

(or perhaps incase updates or python packages eventually stop being able to be installed, for example, via 'sudo apt install python3-numpy' on the released EV3dev Debian 9 "Stretch" armel version...)

--as discussed in #1228, for basic EV3 core set sensors/motors the EV3dev beta "Buster" (which uses python 3.7 and apparently has Brickman 0.10.2 in the default docker pull rather than 0.10.3) does boot, updates via internet, connects to VScode, & appears to run at least basic programs (so far I've just used basic motors & sensors)
https://github.com/ev3dev/docker-library/tree/master/ev3dev-buster
https://github.com/ev3dev/ev3dev/issues/1228#issuecomment-470201945
https://github.com/ev3dev/lego-linux-drivers/compare/ev3dev-buster...Eisverygoodletter:lego-linux-drivers:ev3dev-buster

The EV3dev Beta Debian 10 "Buster" SD card image can be made using Brickstrap (just by following the provided instructions & terminal commands): https://github.com/ev3dev/brickstrap
In my case, I used Ubuntu 22.04 LTS running on WSL2 in Windows 10, & followed the various ev3dev instructions provided on the respective github pages ( = installed docker & brickstrap with apt from repository, pulled the "ev3-buster-generic" docker image & ran brickstrap then copied the .img to a windows folder & flashed a SDcard with Etcher & booted the EV3:

========== ev3dev-sysinfo ==========
Image file:         ev3dev-buster-ev3-generic-2020-04-10
Kernel version:     4.19.85-ev3dev-3-beta2-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165360CCF6
BOARD_INFO_TYPE=main

--also, there is a potentially interesting "user created update" (i.e. tinker & use at your own risk & with caution...) of EV3dev Debian 9 "Stretch" image that has Python 3.9.16 in the discussion at #1558 ...it does boot, etc.: I did succeed at flashing a SD card with the provided 14GB .ima image using the trial version of WinImage & have started tinkering...

https://github.com/ev3dev/ev3dev/issues/1558#issuecomment-1566574102

https://wiki.debian.org/LTS
Debian 8 “Jessie” | i386, amd64, armel and armhf | 17th June 2018 to June 30, 2020
Debian 9 “Stretch” | i386, amd64, armel, armhf and arm64 | July 6, 2020 to June 30, 2022
Current LTS Release(s)
Debian 10 “Buster” | i386, amd64, armhf and arm64 | August 1st, 2022 to June 30th, 2024

https://wiki.debian.org/LTS/Extended
Debian 8 “Jessie” | i386, amd64, armhf, armel | from 2020-07-01 to 2025-06-30
Debian 9 “Stretch” | i386, amd64, armhf | from 2022-07-01 to 2027-06-30
Debian 10 “Buster” | i386, amd64, ...? | from 2024-07-01 to 2029-06-30

Growflavor, 2024-03-26:
@jwaxy @pythoncat1 @atk-r

for tinkering fun in python3 ev3dev2 tinkering with the Debian 11 Bullseye beta2 based on the Kernel version: 4.19.85-ev3dev-3-beta2-ev3 

note that @StepanTheFlowey made a brickman fork that builds on &
works with an apt updated Bullseye beta2 image that is based on the 
original image pulled by: sudo docker pull ev3dev/ev3dev-bullseye-ev3-generic

https://github.com/StepanTheFlowey/brickman/issues/1#issuecomment-2018743067

With the disclaimer & caveat that the following image(s) are "tinker & use for your own 'breaking' fun/learning risk"
(who knows if it may cause CERN to spontaneously form a singularity that consumes your EV3 or... ;-)
since I only use them to help my son learn about VMs & linux, etc...

google drive link to ev3dev11v7.img that can be flashed directly to SDcard by Etcher:

https://drive.google.com/file/d/1uJuOACu0eabON0AuLDwOyCreuW6H-MTB/view?usp=drive_link

this image was built using:

sudo brickstrap create-tar growflavor/ev3images:latest ev3dev11v7.tar

sudo brickstrap create-image ev3dev11v7.tar ev3dev11v7.img

========== ev3dev-sysinfo ==========
Image file: ev3dev-bullseye-ev3-generic-2021-05-10
Kernel version: 4.19.85-ev3dev-3-beta2-ev3
Brickman: 0.11.0+fl0wey1
BogoMIPS: 148.88
Bluetooth: 4.0
Board: board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165360CCF6
BOARD_INFO_TYPE=main

_Note 1_:  using apt is painful onboard the EV3 especially with Bullseye 
make changes/updates in a docker container on a machine with a favorite bricstrap/docker setup, 
then use brickstrap to make a new .img from your updated version to flash with Etcher

_Note 2_: for the moment, sudo docker pull growflavor/ev3images:latest should pull this sha#:

Digest: sha256:e0b53625b2375a4aec942a53860cc2d15f4043693bbc4ed78fa756687c212d76

_Note 3_: no micropython in this image since appears it would need to be rebuilt at least with libffi7 & libmagickwand-6.q16-6 to even have a chance of working with Bullseye

_Note 4_: fun fact "sudo chvt 6 sudo conspy" https://github.com/ev3dev/ev3dev/issues/880#issuecomment-299741628

pythoncat1, 2024-03-26:
@Growflavor Very interesting! Will try it and tinker as soon as I can!

Growflavor, 2024-05-01:
Also, for anyone interested in tinkering with the docker library Stretch image...

sources for backports & security indicated >100 updates available (see below)...

so , here is a link to EV3devDebian9Stretch image based on the EV3dev docker library Stretch image...

- apt update via docker container 
- pip installed RPyC 5.0.0 (which so far seems to work ok even though RPyC 4.x.x is for Python 3.5 -- this just makes it easier to use the same Jupyter Notebook environment that has RPyC 5.x.x with the Stretch, Buster, & Bullseye images) 
- added via apt some python packages like numpy...
- added via apt the beta pybrick-micropython3 package ( #!/usr/bin/env pybricks-micropython3 )

ev3dev9mlv02.zip
https://drive.google.com/file/d/1AsLfR20OCXT4Lsv-pELsOdRfO8hqC90B/view?usp=drive_link

Ev3dev9mlv02.zip sha-256: 
2FBE5EECDDD2A72CE282BAEBAAA4D192C4986270870A36FFCFE48D0934E85838

Ev3dev9mlv02.img sha-256: 
244B3D45D91BA0150442B76461581AD375B91BBF218621E762B3C6DFFF280AB8

snip from the start of the update in the docker container:

robot@faeecdd4302c:~$ sudo -S nano /etc/apt/sources.list
robot@faeecdd4302c:~$ sudo apt update
Ign:1 http://archive.debian.org/debian stretch InRelease
Hit:2 http://archive.debian.org/debian-security stretch/updates InRelease
Hit:3 http://archive.debian.org/debian stretch-backports InRelease
Hit:4 http://archive.debian.org/debian stretch Release
Hit:5 http://archive.ev3dev.org/debian stretch InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
141 packages can be upgraded. Run 'apt list --upgradable' to see them.
robot@faeecdd4302c:~$ apt list --upgradable

pythoncat1, 2024-05-03:
> Also, for anyone interested in tinkering with the docker library Stretch image...
> 
> sources for backports & security indicated >100 updates available (see below)...
> 
> so , here is a link to EV3devDebian9Stretch image based on the EV3dev docker library Stretch image...
> 
> * apt update via docker container
> * pip installed RPyC 5.0.0 (which so far seems to work ok even though RPyC 4.x.x is for Python 3.5 -- this just makes it easier to use the same Jupyter Notebook environment that has RPyC 5.x.x with the Stretch, Buster, & Bullseye images)
> * added via apt some python packages like numpy...
> * added via apt the beta pybrick-micropython3 package ( #!/usr/bin/env pybricks-micropython3 )
> 
> ev3dev9mlv02.zip https://drive.google.com/file/d/1AsLfR20OCXT4Lsv-pELsOdRfO8hqC90B/view?usp=drive_link
> 
> Ev3dev9mlv02.zip sha-256: 2FBE5EECDDD2A72CE282BAEBAAA4D192C4986270870A36FFCFE48D0934E85838
> 
> Ev3dev9mlv02.img sha-256: 244B3D45D91BA0150442B76461581AD375B91BBF218621E762B3C6DFFF280AB8
> 
> After flashing the .img to SD (e.g. via Balena Etcher), on first boot, the boot sequence may pause with a notice in the micro text on the screen that root access is locked. In my case, I press the center button = 'Enter" to continue boot, then reboot from Brickman menu once the 1st boot completes. After the reboot, I then connect to VScode etc.
> 
> ![EV3imageFirstBootDSC08459](https://private-user-images.githubusercontent.com/1580559/327303528-961c10b5-6fb7-4b1a-b1fd-7fc65d975169.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3NDQzOTQsIm5iZiI6MTcxNDc0NDA5NCwicGF0aCI6Ii8xNTgwNTU5LzMyNzMwMzUyOC05NjFjMTBiNS02ZmI3LTRiMWEtYjFmZC03ZmM2NWQ5NzUxNjkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUwMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MDNUMTM0ODE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGVhYmE3ODNkNzIwYmQxMmIxYjIwMTI1MzlhZWYwODQ5YzYxYmVkYTA1NjZmZDVmMjFkM2E0YjMyMDM2MGE4MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.2tbuN9s-oU4af-oIbYeW0rbxUEumhdhk7vNrxvqUjhI)
> 
> snip from the start of the update in the docker container:

Interesting 👀 
Is it possible to use newer versions of Python/Debian packages in this build?

Growflavor, 2024-05-04:
> > Also, for anyone interested in tinkering with the docker library Stretch image...
> > sources for backports & security indicated >100 updates available (see below)...
> > so , here is a link to EV3devDebian9Stretch image based on the EV3dev docker library Stretch image...

> Interesting 👀 Is it possible to use newer versions of Python/Debian packages in this build?

some 'tinker thoughts':

1) the objective of this specific image was primarily to show that the reasonably well documented process to docker pull, adjust in docker container, then use Brickstrap to turn the docker container into a flash-able/bootable EV3 SDcard image still works & is quite easy to do even for a person familiar with computers but not previously experienced specifically with docker/EV3/etc.

2) RE: "Is it possible to use newer versions of Python/Debian packages"

a) you can easily [tinker with 'make altinstall' of Python from source](https://superuser.com/questions/686980/how-to-install-alternative-version-of-python-beside-distro-supplied) in the docker container.  

b) As for 'Debian packages':
       i) if one needed a feature from a newer Debian package, a first step might be to see if that feature was made available via a backport package, then install just that backport & its dependencies via apt:  
                 https://backports.debian.org/Instructions/
                "The co-installability of all available backports is not tested, and it is strongly recommended to opt-into the use of specific backported packages on an as-needed basis."
       ii) if just using the python-ev3dev2 package, one 'tinker' path might be to update via docker an image with either the 'as is' 4.14 or 4.19 armel EV3 kernel to Debian 'Bookworm' then create a venv & pip install python-ev3dev2 -- if no major syntax or methods have been dropped by python 3.11.2 that might 'work'...

3) Overall, in most cases, my sense with the EV3 is that a good practice likely is to use laptops/desktops to do heavy newer python activities & simply exchange low bandwidth commands/data with the EV3 (running a stable image) via any number of the methods python offers..one of the 'beginner accessible' being RPyC.  There are some interesting other examples of other methods such as the LOGO, Pathfind3r, Lidar projects at https://www.ev3dev.org/projects/ as well as Lina's ML for EV3

4) Working with an EV3 image in a docker container does make experimenting much quicker/easier...for example, I see that one Debian maintainer for armel "Bookworm" built a .deb for Python 3.11.2.  So I decided to do a './configure' 'make' 'make altinstall' for python3.11.2 from [a python source package](https://www.python.org/downloads/source/) in this "Stretch" image...so far, I can create a python 3.11 venv but since some headers did not build -- importantly, for example, SSL consequently, so far I can not update or install from pip in that python3.11 venv

excerpt from the ./configure :
checking for --with-openssl-rpath...
checking whether OpenSSL provides required ssl module APIs... no
checking whether OpenSSL provides required hashlib module APIs... no
checking for --with-ssl-default-suites... python
checking for --with-builtin-hashlib-hashes... md5,sha1,sha256,sha512,sha3,blake2
checking for LIBB2... no

<details>
  <summary>example python3.11 altinstall on EV3dev armel "Stretch":</summary>
 
  ```javascript
growflavor@AloEnergia:~$ sudo docker start -ai confident_bohr	
robot@98fc57334730:~$ python --version	
Python 2.7.13	
robot@98fc57334730:~$ python3 --version	
Python 3.5.3	
robot@98fc57334730:~$ python3.11 --version	
Python 3.11.2	
robot@98fc57334730:~$	
	
robot@98fc57334730:~$ source ./py11venv/bin/activate	
(py11venv) robot@98fc57334730:~$ pip --version	
pip 22.3.1 from /home/robot/py11venv/lib/python3.11/site-packages/pip (python 3.11)	
(py11venv) robot@98fc57334730:~$ pip list	
Package    Version	
---------- -------	
pip        22.3.1	
setuptools 65.5.0	
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.	
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping	
WARNING: There was an error checking the latest version of pip.	
(py11venv) robot@98fc57334730:~$	
  ```
</details>

Growflavor, 2024-05-08:
@pythoncat1 & @rayluo RE: 2) b) above...

dist-upgrade to Debian Bookworm via docker container with the branch used to make the ev3dev11v7.img 
based on  @dlech 's Bullseye on 4.19.85 kernel 
actually produced a 'tinker-able' result:

--systemctl output looks decent
--EV3 core set sensors & motors show in @StepanTheFlowey 's Brickman
--RPyC connects via both ethernet & bluetooth
--ev3dev & ev3dev2 modules respond (only tried text to speech & basic motor movement)

Some relevant screenshots in this google photos album:
https://photos.app.goo.gl/Ni4aXNVz2FzsuxQk9

Growflavor, 2024-05-16:
A) @dlech --insights regarding the "Bookworm Kernel" that had activity 2 months ago?
https://github.com/ev3dev/ev3-kernel/tree/ev3dev-bookworm/kernel

B) @rayluo RE: above #3): dlech's prior work comes together nicely in the following Debian "Buster" image:
[image]

<details>
  <summary>Some details about the "Buster" image:</summary>

  ```javascript
ev3dev
ev3dev2
pybricks-micropython: GyroBoy runs OK 
openrobertalab connects

robot@ev3dev:~$ python --version
Python 2.7.16
robot@ev3dev:~$ python3 --version
Python 3.7.3
robot@ev3dev:~$ python3.11 --version
Python 3.11.9
robot@ev3dev:~$ source py11venv/bin/activate
(py11venv) robot@ev3dev:~$ python --version
Python 3.11.9
(py11venv) robot@ev3dev:~$ pip list
Package        Version
-------------- -----------
numpy          1.25.2
pillow         10.3.0
pip            24.0
plumbum        1.8.3
python-ev3dev  1.2.0
python-ev3dev2 2.1.0.post1
rpyc           5.3.1
setuptools     65.5.0
(py11venv) robot@ev3dev:~$ 

note: I suggest stopping the openrobertalab service when using the venv:
sudo systemctl stop openrobertalab

  ```
</details>

https://drive.google.com/file/d/1v47s8ryTqMvyriR02-hL4xMNsCQ9VSgU/view?usp=sharing
ev3dev10img01.zip SHA-256: 4AC95E36C0A684CA5BA22452A0145514F3C133F2955D48E6C6EA8A073C2BBAFF
ev3dev10img01.img SHA-256: 89D5FFD3E67FF77EB491EB23263121FF2F3E601410CF79EC9358CD31D08CC57A

(will update with link to the docker image)
note: the py11 venv pip builds wheels from source...so use docker to add more packages to the venv...

I look forward to hearing from those who try the image if the Py11 ends up actually being useful or not...

dlech, 2024-05-16:
> do you have any insights regarding the "Bookworm Kernel" that had activity 2 months ago?

For my day job, I am doing Linux kernel work full time now, so I was playing around to see how bad it is to update the ev3dev drivers to the latest kernel. It would still take a bit of work to get it working. Don't know  if I would get back to it any time soon.

dlech, 2024-05-16:
By the way, if you want to make any pull requests to get images building on GitHub CI or anything like that, feel free. I'm happy to take a look. I might not say "yes" to everything but it looks like there's lots of useful work being done here.

Growflavor, 2024-05-29:
I've been thinking about @rayluo 's suggestion RE: https://github.com/ev3dev/ev3dev/issues/1626#issuecomment-2046078842

I sense that enabling "github discussions" would be perfect even for just this ongoing conversation...

All of us here have done some nice explorations recently & would be nice to have this more visible than just this 'issue thread' in case others what to try out the images, share issues or successes, make new contributions, etc.

fun note: @StepanTheFlowey has shared .deb for a 4.19 kernel update:
https://github.com/StepanTheFlowey/ev3dev-buildscripts
https://github.com/StepanTheFlowey/ev3-kernel

My son & I plan to be here for at least the next few years learning from what is effectively a wonderful 'practicum' that @dlech  has created with the whole set of EV3dev associated repositories.
  
"If a repository administrator or project maintainer enables [GitHub Discussions](https://docs.github.com/en/discussions/collaborating-with-your-community-using-discussions/about-discussions) for a repository, anyone who has access to the repository can create and participate in discussions for the repository"

thoughts?

dlech, 2024-05-29:
Well, if it make you happy... 😄  

https://github.com/orgs/ev3dev/discussions

Growflavor, 2024-06-18:
> Well, if it make you happy... 😄
> https://github.com/orgs/ev3dev/discussions

Thanks :-)  I created:

Discussion #1634 for posts testing the Python 3.11 venv in the experimental 'Buster' image...[our tests are posted here](https://github.com/orgs/ev3dev/discussions/1634#discussioncomment-9797496) as we do them...

Discussion #1636 for discussion of all the experimental images e.g. Bookworm, Bullseye, Buster, Stretch

Discussion #1633 for 'Best of EV3dev related links'

rayluo, 2024-06-18:
Thanks, @Growflavor , for keep pushing the envelope! I'm currently falling behind but I surely intend to catch up someday!

With regard to testing, I do not have anything specific in mind. Occasionally, I ran into small issues here and there in some Python dependencies, which could have already been fixed in their later versions. The usual and efficient approach would be trying to upgrade and then reproduce the issue, rather than attempting to debug and re-fix an old bug. But `pip install ...` did not work consistently on the old image, so I can't always pull in the latest packages.

My quest for a newer ev3dev wasn't aiming for any specific bugfix, but more on trying to establish a modern foundation so that the entire ev3dev - and Python's pip in my case - can tap into the latest improvements from the ecosystem.

@Growflavor , your work on building newer image for ev3dev probably already achieved that goal. We can all play around with that. And, who knows, someday @dlech might choose to ship a latest image officially. :-)

Bricklou, 2025-05-09:
Hi, any update on the subject?

Growflavor, 2025-05-09:
> Hi, any update on the subject?

I suggest reading through:

Discussion #1636 & Discussion #1661

Discussion #1634 & issue #1635

( and you are welcome to add key items to Discussion #1633 )