ev3dev/ev3dev #1556: EV3dev+PiStorms+Grove?

Aiden-Frank, 2023-02-15:
First, thank you to everyone who has built this software.
I'm trying to use a Grove Voltage Divider with a PiStorms-v2.  I have it plugged in via a grove-to-mindstorms adapter, but I can't get it to understand the input.  Is there a way to read the Grove sensor output?

I am trying to measure voltage across a resistor.  The resistor is flexible and acts as a "whisker" that tells the robot if it's bumping into things while it's moving.

 - OS and version: MacOS 12.2.1
 - Connection Type: Wi-Fi
 - Software: Visual Studio Code

 - Device: PiStorms v2 mounted on Raspberry Pi 3b

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2020-04-10
Kernel version:     4.14.114-ev3dev-2.3.5-rpi2
Brickman:           0.10.3
BogoMIPS:           38.40
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000f73e090b
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.09
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux
```
This is the code I'm trying to run.  Currently, it identifies the Grove as a Touch Sensor emitting a constant 1.

```
#!/usr/bin/env python3
from ev3dev2.display import Display
from ev3dev2.sensor import INPUT_1,INPUT_2, INPUT_3, INPUT_4, Sensor
from ev3dev2.port import LegoPort
from ev3dev2.button import Button 
import time

stop=0

def load_sensor(port,mode): 
    #Function to initiate sensors; port should be INPUT_N where N is 1 to 4
    #mode is a string describing which driver to use (see https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#supported-sensors)
    #device is also a string, see site above
    sensor=LegoPort(port)
    sensor.mode=mode
    time.sleep(0.5)
    return sensor

right_whisker_port=load_sensor("pistorms:BAS2","ev3-analog")
right_whisker=Sensor('pistorms:BAS2')
time.sleep(1)

while stop==0:
    right_value=right_whisker.value()
    print(right_value)
    time.sleep(0.5)
    if Button.enter==True:
        stop=1
```

dlech, 2023-02-15:
I think you might need to set the mode of the port to "nxt-analog" instead of "ev3-analog".

Aiden-Frank, 2023-02-15:
Thank you for responding.  Unfortunately I've tried that already: the output randomly varies from about 5620 to 5650, regardless of whether the sensor is plugged in.

dlech, 2023-02-15:
After looking at [the docs](http://www.mindsensors.com/pdfs/GroveSensorAdapter-User-Guide.pdf), I see the sensor adapter is an I2C device itself and not just a passthrough device. So you will need set the mode of the port to "other-i2c" and use the Python SMBus library to send I2C commands to the adapter.

Aiden-Frank, 2023-02-16:
Thanks so much, I'll try that when I have time this weekend.

Aiden-Frank, 2023-02-19:
I set the mode to that, but I got this error:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 274, in _set_attribute
    attribute.write(value)
OSError: [Errno 22] Invalid argument

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/robot/Snoofer_bot_OS6/whisker_test.py", line 20, in <module>
    right_whisker_port=load_port(INPUT_2,"other-i2c")
  File "/home/robot/Snoofer_bot_OS6/whisker_test.py", line 16, in load_port
    port.mode=mode
  File "/usr/lib/python3/dist-packages/ev3dev2/port.py", line 128, in mode
    self._mode = self.set_attr_string(self._mode, 'mode', value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 337, in set_attr_string
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 277, in _set_attribute
    self._raise_friendly_access_error(ex, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 297, in _raise_friendly_access_error
    driver_error)
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 54, in chain_exception
    raise exception from cause
ValueError: One or more arguments were out of range or invalid, value b'other-i2c'
----------
Exited with error code 1.
```

dlech, 2023-02-19:
Yes, that looks like the correct mode for PiStorms.