ev3dev/ev3dev #1557: Nxt issue

Microtrucker, 2023-02-20:
I am writing this to ask has anyone found a replacement for the lego nxt display. I have tried everything online and it doesn't work....any help is appreciated.

dlech, 2023-02-20:
Reflowing the solder on the capacitors fixed the screen for me. There are many videos on youtube about how to do this.

Microtrucker, 2023-02-20:
> Reflowing the solder on the capacitors fixed the screen for me. There are many videos on youtube about how to do this.

I have tried all these and nothing helped I know the brick its self is good it updated

dlech, 2023-02-20:
Maybe you could find an old dead NXT for parts and use the screen from there? 

Or there are some remote control program that allow viewing the NXT screen remotely that you could use as a substitute for the screen.

Microtrucker, 2023-02-20:
> Maybe you could find an old dead NXT for parts and use the screen from there?
> 
> Or there are some remote control program that allow viewing the NXT screen remotely that you could use as a substitute for the screen.

That's why I am trying to find a new screen that would work

Microtrucker, 2023-03-15:
> I managed to repair a few NXTs by connecting the display via soldered
> wires. Its a bit of work. In some cases I used 0.6 mm tinned steel wire and
> made 12 small clips that I slided onto the glass of the display where the
> contacts are. I completely removed the original cable and the glue so just
> the glass and the very thin metalised contacts remain. I further soldered
> wires onto the clips and connected these to the PCB. It needs some pacient
> to make the clips and that they have the right pressure. Copper wires do
> not work as they are too soft. Finally I fixed the clips with some glue at
> the back side of the display. An alternative is to make a connector that
> slides onto the display contacts. I used a milling machine for that, maybe
> a 3D print would be easier. It turns out the the clips are easier to make.
> When I repair a next one I will make a picture.

Ok cool ty

legorator, 2023-03-15:
just a picture of such a connector and an example of the clip I mentioned, it has to be cutted off from the remaining steel wire. It is usually used as wire for overhead contact lanes for model trains. It works well so far. 

[image]
[image]
[image]

Microtrucker, 2023-03-15:
Hmmmmm ok ty

legorator, 2023-04-05:
I did a video on this... (it is in German but with English subtitles) Maybe this gives some ideas.
[https://www.youtube.com/watch?v=He0JRRf7_ow]