ev3dev/ev3dev #1558: Upgrade Ev3Dev image python to newer version

jackyqiubao, 2023-02-21:
- **ev3dev version:** ev3dev-stretch-ev3-generic-2020-04-10
- **ev3dev-lang-python version:** python3.5.3

Hi,
It looks like current ev3dev stretch image has not been upgraded for a while, is anyone working on upgrading it to newer version of Debian OS and Python, for example, Debian OS 10 and Python3.9  or Python3.11 ?

dlech, 2023-02-21:
Basically same answer as #1554.

jackyqiubao, 2023-03-05:
Hi David,
Thanks for your information!
I tried to use brickstrap to turn buster docker image you published to bootable image, and burn it to my sd card, launched EV3 with the new buster image. Everything is good, except there is not sub menu under bluetooth menu (stretch version has power option under it), and sensor connection cannot be detected. But the motor can be detected. Did I miss anything? 
Thanks!

dlech, 2023-03-06:
It has been a long time since I looked at it so I don't remember what should work and what doesn't. But it was never finished, so it is expected that some things won't work.

ThisBytes5, 2023-05-11:
jackyqiubao, did you get a successful update? I am trying to use a Raspberry PI to communicate with the EV3Dev install, and not having any luck. From what I found the python version has to be same on both machines.
PI - 3.9.2
So I would like to get the EV2 updated as well.

Nikolay-Zagrebin, 2023-05-23:
Hello everyone.
Updated python on ev3dev, to version 3.9.16.
There was a problem calling the ev3dev2 library.
When running scripts, the system does not see the ev3dev2 module, it gives an error
ModuleNotFoundError: No module named 'ev3dev2'

I understand it is necessary to redefine the module calls for the new version of python, but how?

in general, how do I connect the python-ev3dev2 library to the payton 3.9 version?

dlech, 2023-05-23:
The new Python install will have a different site-packages folder so you will need to install python-ev3dev2 there. If the new Python install has a working `pip`, then you can use that to install it.

Nikolay-Zagrebin, 2023-05-23:
> The new Python install will have a different site-packages folder so you will need to install python-ev3dev2 there. If the new Python install has a working `pip`, then you can use that to install it.

David, thank you very much, I'm moving in this direction about now, but having messed up with installing python, I have not been able to restore python-ev3dev2.
Now I repeat everything from the beginning, but it's a long process))
I have all the steps to install a new python takes about a day of pure time)))

Thanks again for confirming my guesses!!!

ThisBytes5, 2023-05-23:
> I have all the steps to install a new python takes about a day of pure time)))

Will you be sharing the image or the steps once you have everything fully working? I'm traveling right now and will be back home on Monday, so anything I can do to assist I would be happy to, just reach out and let me know.

Nikolay-Zagrebin, 2023-05-23:
> Will you be sharing the image or the steps once you have everything fully working? I'm traveling right now and will be back home on Monday, so anything I can do to assist I would be happy to, just reach out and let me know.

Of course, as soon as I install python, I can unload the image.
Now I'm installing 3.10.10, I hope everything will be successful.
Next, I'll write update-alternatives and do the upload, Let's work together on the final image, I'm not so strong in Linux.
Write to me nikolay-zagrebin@yandex.ru

Nikolay-Zagrebin, 2023-05-24:
> * **ev3dev version:** ev3dev-stretch-ev3-generic-2020-04-10
> * **ev3dev-lang-python version:** python3.5.3
>
> Hi, It looks like current ev3dev stretch image has not been upgraded for a while, is anyone [...]

It didn't work with python version 3.10.10, an error with the ssl module came out during compilation.
As a result, when trying to install python-ev3dev2 asks to update pip. But when trying to update the ip, an ssl connection error occurs (((.

in general, I'm going back to the original one and trying to install python 3.9.16

ThisBytes5, 2023-05-24:
> I'm not so strong in Linux.

So the blind leading the blind, I've got enough info of how to move around at the command prompt, but honestly don't really know Linux that well either. Google is my best friend when I'm in the environment, but will definitely give it a go when I'm back home. Currently at a conference 3k miles from home and all my equipment.

Nikolay-Zagrebin, 2023-05-24:
> So the blind leading the blind, I've got enough info of how to move around at the command prompt, but honestly don't really know Linux that well either. Google is my best friend when I'm in the [...]

Good news, when installing python 3.9.16, all modules were compiled and installed without errors!!!
I continue experimenting!! )))

I noticed for interest how long it takes to install python on ev3, it took 5 hours 5 min

Nikolay-Zagrebin, 2023-05-25:
> The new Python install will have a different site-packages folder so you will need to install python-ev3dev2 there. If the new Python install has a working `pip`, then you can use that to install it.

David, can you tell me how to fix it?
I managed to install python 3.9.16, but the problem is that any packages installed via pip are not visible in python.
I think the problem is that pip and python look into different package installation directories

Nikolay-Zagrebin, 2023-05-26:
> Pillow is a binary extension so would have to be compiled when it is installed. I'm guessing the EV3 is probably running out of memory. The only way I know to work around it is to create a binary wheel in a docker container using the `ev3dev/debian-stretch-armel-cross` image and then copy the binary wheel to the EV3 for installation.

David, thank you for your help!!
While I continue experimenting with command parameters.

Nikolay-Zagrebin, 2023-05-29:
That's it, I posted the final version.
Currently installed:
1. Python 3.9.16
2. pip 23.1.2
3. Pillow 8.0.0
4. ev3dev libraries

https://drive.google.com/drive/folders/1THkrEt4HaVb3WftCPcanT-3_YIArbgAp?usp=sharing

Then who needs what, install from pip.
Attention!! It is better not to install software from the repository, if you install something, look carefully so that python 3.5 is not installed!!

I'm experimenting further, my goal is to install the spatialmath library

Nikolay-Zagrebin, 2023-05-30:
> in the previous image post you also posted a doc with the steps on how to duplicate the set up, are you planning on posting the steps again? I honestly just want it for future reference if a newer version is required and you're not around to assist

Posted.
Use the image_1 image, python 3.5 has been removed there, synchronization and update of all software from the Debian repository has been carried out.
Don't install python 3.10.10 - it didn't start for me, maybe of course just the stars didn't converge))).
If you experiment with 3.11 or 3.12, write the results.
Thanks

ThisBytes5, 2023-05-31:
Got the new image set up, able to run the sample motorTest.py ON THE DEVICE with the IP address as 127.0.0.1 as described here:
https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/ev3dev-stretch/rpyc.html

If I change the Ip address to it's external IP (192.168.1.192) it fails with Connection refused
Running from my windows box I get the same basic error:
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

This did not happen on the previous image (at least that I remember).

Being Linux novice I'm not really clear how to resolve this issue. Looking at how to disable firewall on line, the two commands I found (filewalld and ufw) don't exist on the device.

Any suggestions?

Nikolay-Zagrebin, 2023-05-31:
try installing from the Debian repository
sudo apt-get install python3-rpyc

ThisBytes5, 2023-06-01:
> try installing from the Debian repository

Did this last night, wound up downgrading the Python Version to 3.5.x. Will start with a fresh image this weekend and see what I can get working.

Nikolay-Zagrebin, 2023-06-01:
> Did this last night, wound up downgrading the Python Version to 3.5.x. Will start with a fresh image this weekend and see what I can get working.

Yes, that's what I wrote about, now we need to carefully install something from the Debian repository

Nikolay-Zagrebin, 2023-06-12:
> Did this last night, wound up downgrading the Python Version to 3.5.x. Will start with a fresh image this weekend and see what I can get working.

Did you manage to deal with RPyC ?

ThisBytes5, 2023-06-12:
> Did you manage to deal with RPyC ?

No sorry I've not had the time. Wife and kids are gone starting next Monday for the week, so I should have some free time at night that week.

Nikolay-Zagrebin, 2023-06-13:
> No sorry I've not had the time. Wife and kids are gone starting next Monday for the week, so I should have some free time at night that week.

While I was testing the installation of a newer version of python, I looked through the RPyC documentation.
In general, if I run a classic server with the start parameter, then I can access ev3 with Jupyter Notebook.
Run the RPyC server on ev3 using the command

sudo python3 /usr/bin/rpyc_classic.py --host=X.X.X.X

ThisBytes5, 2023-06-19:
> Run the RPyC server on ev3 using the command
> 
> sudo python3 /usr/bin/rpyc_classic.py --host=X.X.X.X

With your image? if so do I need to do any updates on Python, or just the image itself?
x.x.x.x <-- I assume this is the IP address of the ev3?

Nikolay-Zagrebin, 2023-06-19:
> With your image? if so do I need to do any updates on Python, or just the image itself? x.x.x.x <-- I assume this is the IP address of the ev3?
> 
> I should be able to work on this tonight.

Yes, I tested with my own image.
Additionally installed, like you, rpyc

x.x.x.x - yes, this is the IP of your ev3 on your local network

ThisBytes5, 2023-06-23:
I wanted to let you know I have given up on rypc and moved over to PyBricks and just converting the code to use it, and so far it's working great.
https://pybricks.com/install/

Thanks for the assistance, and I hope things work out for you

Nikolay-Zagrebin, 2023-06-23:
> I wanted to let you know I have given up on rypc and moved over to PyBricks and just converting the code to use it, and so far it's working great. https://pybricks.com/install/
> 
> Thanks for the assistance, and I hope things work out for you

OK thanks!! And good luck to you!!
I got so carried away with updating Python that I am now experimenting with installing 3.11 and 3.12)))

Growflavor, 2023-12-09:
> I got so carried away with updating Python that I am now experimenting with installing 3.11 and 3.12)))

@Nikolay-Zagrebin thank you for this interesting thread & image...I am curious what you found regarding python 3.11 & 3.12? :-)

Nikolay-Zagrebin, 2024-06-04:
> > I got so carried away with updating Python that I am now experimenting with installing 3.11 and 3.12)))
> 
> @Nikolay-Zagrebin thank you for this interesting thread & EV3 image with Python 3.9.16 !
> 
> & RE: "...to install the spatialmath library" I am curious, was the EV3 able to handle anything with spatialmath? :-)
> 
> since the current thread mentions [RPyC on ev3dev](https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/ev3dev-stretch/rpyc.html) (e.g. "run a python program on your laptop that controls an ev3dev device. This can be useful if your robot requires CPU intensive code that would be slow to run on the EV3.")
> 
> for anyone curious, three other useful computation/simulation related EV3 references:
> 
> a) Lina does not use RPyC but still interesting how her µanthánō uses docker containers to explore & do more cpu intensive ML training computations: Lina Peters' intriguing [EV3 ML digit classification masters thesis](https://www.researchgate.net/publication/337868874_An_Educational_Programming_Environment_for_Teaching_the_Principles_of_Machine_Learning_using_LEGO_MINDSTORMS): µanthánō (docker based 'Blocky' ML tool to help with the ML training that the EV3 takes 'too long to do onboard') : https://github.com/leenahere/manthano project overview: https://leenahere.github.io/blog/ml-with-ev3/ basic line follower digit classification: https://github.com/leenahere/ai-for-ev3 a fork for 'letter classification': https://github.com/MariosMarinos/ai-for-ev3
> 
> b) [ev3dev-lang-python-fast](https://github.com/QuirkyCort/ev3dev-lang-python-fast)
> 
> c) Useful for young students of python/robotics: GEARS EV3 simulator & EV3 Blocky-->Python: https://quirkycort.github.io/gears/public/
At the moment, I have not reached spatialmath, I am stuck with installing numpy))
But along the way I managed to install python version 3.10.14 on my mind3, I am just finishing the tests.

Growflavor, 2024-06-17:
> At the moment, I have not reached spatialmath, I am stuck with installing numpy)) But along the way I managed to install python version 3.10.14 on my ev3, I am just finishing the tests.

Great!  Thanks for the update :-)

At Discussion #1634 I have posted a Debian 10 "Buster" image (system python 3.7) that also has an altinstall Python 3.11 venv.  I included a link to its docker container so that if you have a docker setup that more packages for the python 3.11 venv can be built from source by pip (so far, I only had pip make numpy, rpyc & the ev3dev/2 packages).

Nikolay-Zagrebin, 2024-06-17:
> Great! Thanks for the update :-)
>
> At Discussion #1634 I have posted a Debian 10 "Buster" image (system python 3.7) that also has an altinstall Python 3.11 venv. I included a link to its docker [...]

I'm following your posts, thank you so much for your work. Everything is very interesting.
I decided to build my image on Bullseye, but I can't install Brickman 0.11 from StepanTheFlowey correctly.

I do everything according to the instructions, the whole installation goes fine, but after flashing the SD card, the EV3 itself has an interface with unreadable text (((
Is there anything else undocumented?

There is a desire to test approximately similar builds on Bullseye and Stretch

Growflavor, 2024-06-17:
RE: 'I decided to build my image on Bullseye, but I can't install Brickman 0.11 from StepanTheFlowey correctly.
I do everything according to the instructions, the whole installation goes fine, but after flashing the SD card, the EV3 itself has an interface with unreadable text ((('
& 'Maybe at least one hint that the fonts were installed?' at #1634 https://github.com/orgs/ev3dev/discussions/1634#discussioncomment-9798131

Below is a .txt with the apt list --installed for the ev3dev11v7 image provided at:

https://github.com/ev3dev/ev3dev/issues/1554#issuecomment-2019205120

compare it with the apt list -installed from the image that is having trouble...

[AptListInstalled_ev3dev11v7.txt](https://github.com/user-attachments/files/15876776/AptListInstalled_ev3dev11v7.txt)