ev3dev/ev3dev #1562: Running C++ binarys on the brick ('GLIBC_2.27' not found)

Eric-Bwr, 2023-03-27:
Hello, I am trying to follow the C++ Tutorial listed on the ev3dev.org site.

Step 7 describes how to install gcc-5. 
-> https://github.com/ddemidov/ev3dev-lang-cpp

So I add the three lines to the source list which now looks as follows:
```
deb http://httpredir.debian.org/debian stretch main contrib non-free
#deb-src http://httpredir.debian.org/debian stretch main contrib non-free

deb http://security.debian.org/ stretch/updates main contrib non-free
#deb-src http://security.debian.org/ stretch/updates main contrib non-free

deb http://archive.ev3dev.org/debian stretch main
#deb-src http://archive.ev3dev.org/debian stretch main

deb http://httpredir.debian.org/debian unstable main contrib non-free
deb http://security.debian.org/ unstable/updates main contrib non-free
deb http://archive.ev3dev.org/debian unstable main
```

After which I run `sudo apt update`

Which gives:
```
Hit:1 http://security.debian.org stretch/updates InRelease                                                                          
Ign:2 http://httpredir.debian.org/debian stretch InRelease                                                                          
Get:3 http://httpredir.debian.org/debian unstable InRelease [180 kB]                      
Ign:4 http://security.debian.org unstable/updates InRelease                               
Hit:5 http://archive.ev3dev.org/debian stretch InRelease                                  
Ign:6 http://archive.ev3dev.org/debian unstable InRelease                                 
Err:7 http://security.debian.org unstable/updates Release                                 
  404  Not Found
Hit:8 http://httpredir.debian.org/debian stretch Release                 
Err:9 http://archive.ev3dev.org/debian unstable Release                  
  404  Not Found
Get:10 http://httpredir.debian.org/debian unstable/contrib armel Packages [42.4 kB]                                                                                                                          
Get:11 http://httpredir.debian.org/debian unstable/non-free armel Packages [57.7 kB]                                                                                                                         
Get:12 http://httpredir.debian.org/debian unstable/main armel Packages [8,985 kB]                                                                                                                            
Reading package lists... 80% 
```

But it seems to be stuck at 80%. I let it run for hours. Nothing happens.
The SSH-Connection is also not broken (im using wired) as I could abort it easily.
It seems to me that something in some repository got bigger or changed or is not available (as the 404 above suggest) and thus it fails to read some packages or something.
I am not that familiar with aptly. I know the basics.

Oh and im trying to fix this issue:
/lib/arm-linux-gnueabi/libm.so.6: version `GLIBC_2.27' not found
Which occurs when I try to run the demos (for. ex. drive-test)
That obviously makes sense though because I cant even get past step 7.

Im sure im just stupid and someone else can figure it out immediatly.

Greetings and thank you in advance,
Eric

dlech, 2023-03-27:
See https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/ for the recommended way to cross-compile for ev3dev.

dlech, 2023-03-27:
Alternately, you can statically link your C++ program and it should work.

Eric-Bwr, 2023-08-18:
@dlech 
If you are using a Linux host you must have the proper format registered with binfmt_misc.
Can you be a bit more specific whats meant with this? I tried to set some stuff I found online but that didnt really work at all...
I got docker and everything set up and compiling but it fails when I try to run the example code (hello.c) in the docker container.

dlech, 2023-08-18:
https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

> Can you be a bit more specific whats meant with this?

See note in link above about `qemu-user-static`.