ev3dev/ev3dev #1564: EV3 Mindstorm Problems with sudo apt get/install etc. connection breaks

nobbi9999, 2023-04-05:
Hello,
everything works really good, but i have following problem.
I want to install portaudio with  sudo apt-get install libasound-dev

The Downloading starts fine but after a few minutes (74 % finished) the SSH/Wifi-Connection breaks
I tried it several times....rebooting .... etc.

I think there is general problem with my sudo-Command, all my sudo command fail after a few minutes, e.g.

sudo apt install yum
sudo apt-get-update

Its not a syntax Problem, the command starts, but the Connection breaks always  e.g.  70 % - 80 %, not finished
Perhaps its' a timeout problem ? the Value is 30 an the break down is after ca. 10 minutes

Any ideas ? Thank you in advance

dlech, 2023-04-05:
I would try USB instead of Wi-Fi. Also, if you have a slow SD card, these commands can take a long time.

nobbi9999, 2023-04-05:
Hallo David,i don't know if it possible to Use USB with the Brickmann Operation System. I cant find a USB Connection entry under Networks and Connecticut, only Bluetooth and WiFi on the brickhow can i set USB?--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

dlech, 2023-04-05:
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

nobbi9999, 2023-04-05:
Thank You very much......--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

   
  https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/