ev3dev/ev3dev #1567: HiTechnic Accelerometer

mcdhub15, 2023-05-07:
How does one  parse the data coming IN from the HiTechnic Accelerometer?   I need th LOG the x, x and z positions as the robot moves around.   Thje code below is working on the EV3 but the text printing out to the LCD is very tiny.

#!/usr/bin/env python3

from ev3dev2.motor import *
from ev3dev2.sensor import *
from ev3dev2.sensor.lego import *
from ev3dev2.sound import Sound
from ev3dev2.button import *
import time

#from ev3dev2.sensor.lego import Accelerometer

#led library
from ev3dev2.led import Leds

'''
#sensor Libraries
from ev3dev2.sensor import *
from ev3dev2.sensor import *
from ev3dev2.sensor.lego import*

#Motor Library
from ev3dev2.motor import *

'''

# HiTechnic Accelerometer for NXT
#  https://github.com/ev3dev/ev3dev-lang-python/issues/596
from ev3dev2.sensor import *
a = Sensor(address=INPUT_1, driver_name='ht-nxt-accel')
#debug_print(c.address)
#value = c.value
#debug_print(value)

us = UltrasonicSensor(INPUT_3)
#ts = TouchSensor(INPUT_1)
#a = Accelerometer(INPUT_1)
#cl = ColorSensor('in4')
sound = Sound()
leds = Leds()

lm = LargeMotor(OUTPUT_A)

sound.speak("Let us use the accelerometer")

leds.animate_rainbow()
running = True
while(running):

    dist_cm = us.distance_centimeters
    print(dist_cm)

    #print(a.value())

    value = a.value
    print(value)
    # got to get x, y and z

    #loc = a.get_acceleration()
    #print(loc.x, loc.y, loc.z)

    if (dist_cm < 15):
        print ("Time to run the motor")
        lm.run_timed(time_sp=500, speed_sp=720)

'''
    pressed = ts.is_pressed

    if pressed == True:
        print("TRUE")
    else:
        print("FALSE")

    if pressed == True:

'''

How do you ...?

**Background Information**

I am trying to ... because I want to build a robot that ...

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2023-05-07:
It looks like you want the `ALL` mode.

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-accel

Also, you should be able to use `brickrun -r ./my-script.py` in a remote terminal to print to the remote terminal instead of on the screen of the EV3.

mcdhub15, 2023-05-08:
Hi David,

I will try 'lego-nxt-all' to get at the x,y and z values.

Also, trying to get the Lego NXT Sound Sensor to work with the EV3 robot:

#!/usr/bin/env python3

from ev3dev2.motor import *
from ev3dev2.sensor import *
from ev3dev2.sensor.lego import *
from ev3dev2.sound import Sound
from ev3dev2.button import *
import time

#from ev3dev2.sensor.lego import Accelerometer

#from ev3dev2.sensor.lego import SoundSensor

#led library
from ev3dev2.led import Leds

'''
#sensor Libraries
from ev3dev2.sensor import *
from ev3dev2.sensor import *
from ev3dev2.sensor.lego import*

#Motor Library
from ev3dev2.motor import *

'''

# Lego NXT Sound Sensor
#
https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#lego-nxt-sound
# import - no idea   from ev3dev2.sensor import *

#s = Sensor(address=INPUT_2, driver_name='lego-nxt-sound')
#debug_print(c.address)
#db = s.value
#debug_print(value)

from ev3dev2.sensor.lego import SoundSensor

from ev3dev2.sensor import INPUT_4

from ev3dev2.port import LegoPort

from time import sleep

p = LegoPort(INPUT_4)

p.set_device = 'lego-nxt-sound'

sleep(1)

s = SoundSensor()
MODE_DB = 'DB'

# HiTechnic Accelerometer for NXT
#  https://github.com/ev3dev/ev3dev-lang-python/issues/596
from ev3dev2.sensor import *
#a = Sensor(address=INPUT_1, driver_name='ht-nxt-accel')
a = Sensor(address=INPUT_1, driver_name='ht-nxt-accel')
#debug_print(c.address)
#value = a.value
#debug_print(value)

us = UltrasonicSensor(INPUT_3)
#ts = TouchSensor(INPUT_1)
#a = Accelerometer(INPUT_1)
#cl = ColorSensor('in4')
sound = Sound()
leds = Leds()

lm = LargeMotor(OUTPUT_A)

sound.speak("Let us use the accelerometer")

leds.animate_rainbow()
running = True
while(running):

    dist_cm = us.distance_centimeters
    print(dist_cm)

    #print(a.value())

    value = a.value
    print(value)
    #print("X:" + str(value.x))

    # got to get x, y and

    #db = s.value
    #print(db)

    #if db > 400:
    #    sound.play_tone
    #   sound.speak("I hear you")
    #    time.sleep(2)

    db = s.sound_pressure

    print(db)
    time.sleep(1)

    if db > 200:
        sound.play_song((('D4', 'e3'),('D4', 'e3'),('D4', 'e3'),('G4', 'h'
),('D5', 'h')))

        time.sleep(3)
        lm.run_timed(time_sp=500, speed_sp=300)

    #loc = a.get_acceleration()
    #print(loc.x, loc.y, loc.z)

    if (dist_cm < 15):
        print ("Time to run the motor")
        lm.run_timed(time_sp=500, speed_sp=720)

'''
    pressed = ts.is_pressed

    if pressed == True:
        print("TRUE")
    else:
        print("FALSE")

    if pressed == True:

'''

I don't understand the flat units that are being measured by the sound
sensor. I think the code works
but it only works with values 0 to 90. I was hoping to read values 300 to
1000.

Kiley

Kiley McDaniel
Curriculum Chair (Computer/Business/Technology)
St. Stephen Catholic Secondary School
Bowmanville, Ontario, CANADA
http://www.sscss.ca
P: 905-623-3990 x 570
F: 905-623-9991
***@***.***

-- 

Please visit our website at http://www.pvnccdsb.on.ca 
<http://www.pvnccdsb.on.ca> or join us on social media:
Facebook: 
https://www.facebook.com/pvnccdsb <https://www.facebook.com/pvnccdsb>
Twitter: https://twitter.com/pvnccdsb <https://twitter.com/pvnccdsb>
Youtube: https://www.youtube.com/pvncc <https://www.youtube.com/pvncc>

PVNCCDSB IMPORTANT NOTICE: This information is intended only for the use of 
the individual or entity to which it is addressed and may contain 
information that is privileged, confidential and exempt from disclosure 
under the Municipal Freedom of Information and Protection of Privacy Act. 
If the reader of this message is not the intended recipient or the employee 
or agent responsible for delivering the message to the intended recipient, 
you are hereby notified that any dissemination, distribution or copying of 
this communication is strictly prohibited. If you received this 
transmission in error, please notify the sender immediately and then 
permanently delete this message.

dlech, 2023-05-08:
> I will try 'lego-nxt-all' to get at the x,y and z values.

I think you mean this?

```python
a = Sensor(address=INPUT_1, driver_name='ht-nxt-accel')
a.mode = 'ALL'
```

> I don't understand the flat units that are being measured by the sound sensor. I think the code works but it only works with values 0 to 90. I was hoping to read values 300 to 1000.

The measured values are 0 to 100% so getting 0 to 90 sounds reasonable.

dlech, 2023-05-08:
> I wonder of the input from the Hitechnic accelerometer is a list or array that needs a INDEX to access the x, y and z values

The docs I linked in https://github.com/ev3dev/ev3dev/issues/1567#issuecomment-1537553179 explain the data format. I would suggest using the `bin_data()` method to get everything at once instead of `value()`.

```python
x_msb, y_msb, z_msb, x_lsb, y_lsb, z_lsb = a.bin_data("<6b")
x = (x_msb << 2) | (x_lsb >> 6) # no idea if this is right, see footnote in docs
...
```

```python
# or just use msb if last two bits are not important
x, y, z, *_ = a.bin_data("<6b")
```