ev3dev/ev3dev #1568: Wanting to play a video on EV3 Screen

CyanCheetah, 2023-05-17:
**Question**

How do you Play a video on EV3 Screen?

**Background Information**
I have been trying to play a video on the ev3 screen but am struggling to do it. I tried installing OMXplayer but that for some reason does not install properly, same with vlcplayer or pygame. I got fbi to work but I cannot play a gif for some reason, only display jpg's. I made it display a bunch of jpg's in quick succession so that it simulates a stop-motion video but I keep getting black frames between the images as well. So I am trying to make the ev3 screenplay a video. Thanks!

**Desktop (please complete the following information):**
 - OS and version: Windows 11
 - Connection Type: USB
 - Software: VS Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: Latest One

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

CyanCheetah, 2023-05-17:
I even tried mplayer but get this instead
[image]

dlech, 2023-05-17:
You need to run programs with `brickrun` to run them on the screen of the EV3, otherwise it will try to run in a remote terminal and can't access the screen.

CyanCheetah, 2023-05-17:
Thanks! But when I run a brickrun -- /usr/bin/mplayer /home/robot/ev3/videoname.mp4 I get a blank screen. Im sorry I am very new to this haha!

CyanCheetah, 2023-05-17:
am I doing a command wrong or

dlech, 2023-05-18:
You might need some more options like `-vo fbdev /dev/fb0 -framedrop`. If I ever got it to work, it was a very long time ago.

CyanCheetah, 2023-05-18:
thanks! I got the audio to play and the video to play but the video plays after the audio plays. If not thats fine but am curious to see if there is a solution to it

dlech, 2023-05-18:
I would recommend encoding the audio as a 22050Hz mono WAV so no conversion is required during playback to reduce CPU usage. No idea if that will help with the video not playing until later.

CyanCheetah, 2023-05-18:
thanks! Now one question is how do I encode the audio as the mono WAV? Also thank you for your quick responses!! this has been a real help to me!

dlech, 2023-05-18:
This is what ChatGPT said:

> To encode audio to a 22050Hz mono WAV file using ffmpeg, you can use the following command:
> 
> ```shell
> ffmpeg -i input_file.mp3 -ac 1 -ar 22050 output_file.wav
> ```
> 
> This command assumes that the input file is an MP3 file named `input_file.mp3` and you want to convert it to a mono WAV file with a sample rate of 22050Hz named `output_file.wav`.
> 
> Here's a breakdown of the command options used:
> 
> - `-i input_file.mp3`: Specifies the input file.
> - `-ac 1`: Sets the number of audio channels to 1, which means mono audio.
> - `-ar 22050`: Sets the sample rate to 22050Hz.
> 
> After running the command, ffmpeg will transcode the input audio file to the specified output format and sample rate.

Then to combine it with the video:

> Yes, you can copy the audio stream without re-encoding by using the `-c:a copy` option in the ffmpeg command. This will preserve the original audio format and quality.
> 
> Here's an example command that copies the audio stream without re-encoding:
> 
> ```shell
> ffmpeg -i input_video.mp4 -i input_audio.wav -c:v copy -c:a copy output_file.mp4
> ```
> 
> In this command, the `-c:a copy` option tells ffmpeg to copy the audio stream as is, without any re-encoding. The `-c:v copy` option is used to copy the video stream without re-encoding as well. The resulting output file, `output_file.mp4`, will contain the original audio and video streams combined into a single file.

CyanCheetah, 2023-05-18:
dang I did all that and get
`
           ************************************************
           **** Your system is too SLOW to play this!  ****
           ************************************************

Possible reasons, problems, workarounds:
- Most common: broken/buggy _audio_ driver
  - Try -ao sdl or use the OSS emulation of ALSA.
  - Experiment with different values for -autosync, 30 is a good start.
- Slow video output
  - Try a different -vo driver (-vo help for a list) or try -framedrop!
- Slow CPU
  - Don't try to play a big DVD/DivX on a slow CPU! Try some of the lavdopts,
    e.g. -vfm ffmpeg -lavdopts lowres=1:fast:skiploopfilter=all.
- Broken file
  - Try various combinations of -nobps -ni -forceidx -mc 0.
- Slow media (NFS/SMB mounts, DVD, VCD etc)
  - Try -cache 8192.
- Are you using -cache to play a non-interleaved AVI file?
  - Try -nocache.
Read DOCS/HTML/en/video.html for tuning/speedup tips.
If none of this helps you, read DOCS/HTML/en/bugreports.html.
`
How unfortunate. would setting the sample rate lower help ease the cpu load?

dlech, 2023-05-18:
Keep playing around with different encodings until you find something that works, I guess.

CyanCheetah, 2023-05-19:
I tried some of the tips that they gave and I got it to play the video and music at the same time... but it is super slow. By that I mean the audio and video are running at like 5 frames per second so the audio is really slow.

CyanCheetah, 2023-05-19:
brickrun -- mplayer /home/robot/ev3/BadApple.mp4 -ao sdl -vo fbdev2:/dev/fb0 -autosync 5 -vfm ffmpeg -lavdopts lowres=1:fast:skiploopfilter=all -cache 16000 -nocache

Heres the command I ran

CyanCheetah, 2023-05-20:
@dlech I actually found a solution while messing around for like 10 hours haha. What I did was figured out mplayer supports mpg and converted the mp4 file to mpg while lowering the audio and video bitrate, and changing it to mono audio, I think I changed audio codec to h264 as well. Oh and lowered fps as well. Thanks tho you saved me with specifying the video output!!!! Btw I appreciate all the help you have given me! It took a while though if I am being honest

dlech, 2023-05-20:
Rock on.