ev3dev/ev3dev #1569: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty

f1refa11, 2023-05-20:
**Describe the bug**
I wanted to try running Minecraft Server(paper) for version 1.8.8, but when i tried to run it, the server started downloading `vanilla jar` and thrown an error:
```
Downloading vanilla jar...
Failed to download vanilla jar
javax.net.ssl.SSLException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
        at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:133)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:320)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:263)
        at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:258)
        at java.base/sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1313)
        at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:408)
        at java.base/sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:567)
        at java.base/sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1587)
        at java.base/sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1515)
        at java.base/sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:250)
        at java.base/java.net.URL.openStream(URL.java:1140)
        at io.papermc.paperclip.Paperclip.checkVanillaJar(Paperclip.java:191)
        at io.papermc.paperclip.Paperclip.checkPaperJar(Paperclip.java:122)
        at io.papermc.paperclip.Paperclip.setupEnv(Paperclip.java:95)
        at io.papermc.paperclip.Paperclip.main(Paperclip.java:49)
Caused by: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
        at java.base/sun.security.validator.PKIXValidator.<init>(PKIXValidator.java:102)
        at java.base/sun.security.validator.Validator.getInstance(Validator.java:181)
        at java.base/sun.security.ssl.X509TrustManagerImpl.getValidator(X509TrustManagerImpl.java:300)
        at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrustedInit(X509TrustManagerImpl.java:176)
        at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:189)
        at java.base/sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:129)
        at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.checkServerCerts(CertificateMessage.java:629)
        at java.base/sun.security.ssl.CertificateStatus$CertificateStatusConsumer.consume(CertificateStatus.java:295)
        at java.base/sun.security.ssl.SSLHandshake.consume(SSLHandshake.java:392)
        at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:443)
        at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:421)
        at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:177)
        at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:164)
        at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1151)
        at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1062)
        at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:402)
        ... 10 more
Caused by: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
        at java.base/java.security.cert.PKIXParameters.setTrustAnchors(PKIXParameters.java:200)
        at java.base/java.security.cert.PKIXParameters.<init>(PKIXParameters.java:120)
        at java.base/java.security.cert.PKIXBuilderParameters.<init>(PKIXBuilderParameters.java:104)
        at java.base/sun.security.validator.PKIXValidator.<init>(PKIXValidator.java:99)
        ... 25 more
```

**To Reproduce**
Steps to reproduce the behavior:
1. `curl https://api.papermc.io/v2/projects/paper/versions/1.8.8/builds/445/downloads/paper-1.8.8-445.jar --output paper.jar` to download the `.jar` file
2. `java -jar paper.jar` to run the server
3. Wait some time and see the error.

**Expected behavior**
```
Downloading vanilla jar...
Patching vanilla jar...
Loading libraries, please wait...
[18:43:39 INFO]: Starting minecraft server version 1.8.8
[18:43:39 INFO]: Loading properties
[18:43:39 WARN]: server.properties does not exist
[18:43:39 INFO]: Generating new properties file
[18:43:39 WARN]: Failed to load eula.txt
[18:43:39 INFO]: You need to agree to the EULA in order to run the server. Go to eula.txt for more info.
[18:43:39 INFO]: Stopping server
>
```

**Desktop (please complete the following information):**
 - OS: [e.g. Window 10, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Other software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534F237C
BOARD_INFO_TYPE=main
```

**Additional context**
None.

f1refa11, 2023-05-20:
Tried [this](https://github.com/PaperMC/Paper/issues/1271#issuecomment-407617533):
```
sudo rm /etc/ssl/certs/java/cacerts
sudo update-ca-certificates -f
```
Still doesn't work.

f1refa11, 2023-05-20:
Installing openjdk-8-jre-headless doesn't help.

dlech, 2023-05-20:
Can you download the jar to you computer then transfer it to your EV3? ev3dev-stretch probably has too old of SSL support.

f1refa11, 2023-06-24:
> Can you download the jar to you computer then transfer it to your EV3? ev3dev-stretch probably has too old of SSL support.

Okay, i'll try it. Anyway, thanks for tech-support!

closed 2023-06-24 by f1refa11