ev3dev/ev3dev #1571: Logitech C270 is not offering any MJPEG format

fwillo, 2023-06-02:
**Describe the bug**
I try bringing up a Logitech C270 on an EV3 but `v4l2-ctl -d /dev/video0 --list-formats-ext` is only showing YUV in a very low resolution. On other Linux devices (and Windows) I'm able to use the whole functionality.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Logitech C270 to the USB 1.1 Port
2. `v4l2-ctl -d /dev/video0 --list-formats-ext`
3. Get
```
robot@ev3dev:~$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 160x120
                        Interval: Discrete 0.100s (10.000 fps)
```

**Expected behavior**
The driver offering more than YUV in poor resolution and framerate

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev 4.14.117-ev3dev-2.3.5-ev3

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=
BOARD_INFO_TYPE=main
```

**Additional context**
I've seen a discussion at #230 where @pabr (https://github.com/ev3dev/ev3dev/issues/230#issuecomment-68692703) suggested to alter ev3dev-kernel/drivers/media/usb/uvc/uvc_video.c and recompile the kernel module, as stated in his reference. I tried this. Modified  uvc_video.c, compiled the kernel with unique name, booted it and ensured my modification was booted. However, I end up with the same result. Is USB 1.1 not offering more bandwidth in order to use MJPEG? USB 2.0 seems to be the Ansatz then. But as in #1550, it seems this more difficult than thought.

I'm a bit puzzled because I find blog posts with ev3dev where people also have the Logitech C270 and get MJPEG ([here](http://www.pabr.org/bricks/brickfpv/brickfpv.en.html#), [here](https://siyahodacom.wordpress.com/2017/10/01/adding-webcam-to-ev3-ev3dev/) and [here](https://ofalcao.pt/blog/series/ad4m4st0r-um-rover-lego)). What am I doing wrong? Should I compile `ev3-buster` from the git? Looking forward to your answers.

fwillo, 2023-06-02:
Thank you for your fast answer. So it seems I have no chance getting my current setup running with a MJPEG stream?

Would it work with the USB 2.0 mini USB connector? My question regards #230 #1550, where no further hints are given if people already got it running.

Can you recommend a camera which reliably offers the possibility of connecting it with MJPEG, good resolution and good framerates?

dlech, 2023-06-02:
What are you trying to accomplish?

The EV3 is old and slow, so depending on what your goal is, I would probably recommend doing image processing somewhere else and send the results to the EV3.

fwillo, 2023-06-02:
I try to construct something which is controllable remotely and returns a video feed back to the remote terminal. The prototype is working. However, the very poor resolution and framerate is improvable. Indeed, the hardware is old and slow and, of course, a solution with an RPi would be preferable. It has to be something working with Mindstorms ecosystem parts and due to budget constraints RPis + modules working with said ecosystem are too expensive. I have to work out a solution with the parts I got.