ev3dev/ev3dev #1572: Lego Mindstorms EV3 stuck at first boot

Jihyeon-Je, 2023-06-07:
I am having issues with the very first boot of ev3dev on Lego Mindstorms EV3. I am using a 32GB microSDHC card and etched the jessie version (using balena etcher). I saw other closed tickets where it's stuck at boot, but in my case it's just stuck at the very first display screen where it just shows "starting...". (I waited 10+ minutes and it won't move). The documentation says I should see two boots- one for mindstorms and one for ev3dev - but it's stuck at the first mindstorms boot with the red LED on. When I unplug the SD card it boots fine. 
I tried re-etching the SD card and tried with different EV3 bricks but I'm still getting the same issue. How can I fix this?

dlech, 2023-06-07:
I would recommend the stretch version instead of the jessie version (unless you have a good reason to run the older version).

Jihyeon-Je, 2023-06-07:
I tried with the stretch version and I'm having the same issue where it just gets stuck at the starting screen. :/

dlech, 2023-06-07:
Do you have a different microSD card you could try?

Kubkubs3234, 2023-07-18:
I have the same issue, i have tried on 4gb and 8gb cards

dlech, 2023-07-18:
It sounds like Etcher has confirmed that both SD cards are defective. I highly recommend using SanDisk Extreme microSDHC. 32GB ones are still in production.

Kubkubs3234, 2023-07-22:
I remember doing this a few years ago and it worked.