ev3dev/ev3dev #1573: MemoryError: memory allocation failed

JonathanDrukker, 2023-06-26:
I am getting the `MemoryError: memory allocation failed, allocating 16 bytes` error when trying to import a large variable
in micropython.
Here is the code:
```
#!/usr/bin/env micropython

from gc import collect; collect()
from paths import path # path is the large varible
```
When using python3 it doesn't happen( `#!/usr/bin/env python3`)
I tried using the pybricks project (which also uses micropython) and it works there too.

Does anyone know a fix? I dug into the source code of both pybricks and the ev3dev built in micropython, and I saw that pybricks has the latest version of micropython, while the ev3dev one has a very old version.
I think that updating the version of micropython will do the trick, but when inspecting the commits on the ev3dev micropython I saw there were a lot of changes that I didn't know how to replicate.

Does anyone know a fix or how to update micropython? Thanks

OS and version: Windows 11 Home
Connection Type: USB
 Software: VS Code

Device: Lego EV3 Brick

```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           227.32
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=0016538379FC
BOARD_INFO_TYPE=main

```

dlech, 2023-06-26:
I don't think updating MicroPython will make much difference since memory allocation and the garbage collector haven't really change much.

Pybricks has a larger default heap. You can use the `-X heapsize=16k` option to increase the heap size of the standard MicroPython. However, you can't do this in the shebang, so you would need to create a wrapper script or something like that.

JonathanDrukker, 2023-06-26:
Can you please elaborate on the wrapper script?

dlech, 2023-06-26:
Something like this?

File: `/home/robot/micropython`

```sh
#!/bin/sh

exec micropython -X heapsize=16K "$@"
```

Make executable: `chmod +x /home/robot/micropython`.

Change your program to

```python
#!/home/robot/micropython

...
```

JonathanDrukker, 2023-06-27:
I am not sure why, but it didn't work for me at the start, but when I changed from `16K` to `16000K` it worked. I think its to do with the function units...
Thank you very much!

Anyone trying to fix the same issue, here is how I did it:
A file called `micropython` , the path to it should be: `#!/home/robot/micropython`:
```
#!/bin/sh
exec micropython -X heapsize=16000K "$@"
```
Then main.py:
```
#!/home/robot/micropython

# code...
```
You might need to call the following function to make the  `micropython` an executable, but for me `micropython` was automatically an executable:
`chmod +x /home/robot/micropython`

If the error still ocures, try playing with the 16000K number, as it represents the heap size (the heap is basically the place were the micropython stores all variable data).

closed 2023-06-27 by JonathanDrukker

dlech, 2023-06-27:
> I changed from `16K` to `16000K`

Oops. Yes, 16K is plenty for other LEGO hubs that aren't running Linux but not very much for the EV3. You could simplify it to `16M` to get 16 megabytes which IIRC is the default used in Pybricks.