ev3dev/ev3dev #1574: How can I double Encoder resoulution?

JonathanDrukker, 2023-06-30:
I have read on Wikipedia a "hack" to double the EV3's motor encoder resolution by enabling `edge triggered interrupts`.
It was done on a GitHub repository: @EAVR/EV3.14, but as per my understanding, it was done via a Raspberry Pi?

Link to Wikipedia (Enhancements section): https://en.wikipedia.org/wiki/Lego_Mindstorms_EV3
Link to EV3.14: https://github.com/EAVR/EV3.14

I looked into the ev3dev files but found no option/setting to enable edge-triggered interrupts.
Might someone know how to do so? Thank you in advance!

OS and version: Windows 11 Home
Connection Type: USB
Software: VS Code

Device: Lego EV3 Brick

```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           227.32
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=0016538379FC
BOARD_INFO_TYPE=main
```

dlech, 2023-07-01:
The necessary code is already in place. Instead of using the `tacho-motor` sysfs interface, you just need to read the count from the `iio` subsystem instead. For example `/sys/bus/iio/devices/iio:device1/in_count0_raw`.

If you are interested in the code, it here: https://github.com/ev3dev/ev3dev-pru-firmware/blob/master/ev3-pru1/main.c

Since it is running in the PRU, we actually just poll the GPIOs of the motor encoders instead of using interrupts.

JonathanDrukker, 2023-07-26:
Hey, I tried analyzing the code, but I am unfamiliar with C code and don't understand what it means.
1. I managed to open the file in Python and read it VIA the built-in `open` function, but as per my knowledge of Python and how slow it is, I `think` it's a suppose a bad approach to do so. Is there a quicker way to do so? Maybe VIA a terminal function?

2. Is the file like in_frequency0_input used to get the velocity? Because from my testing, it seems like it.

3. Can I make the motor move by editing one of the files and giving it a certain frequency or voltage?

4. I tried a similar approach to get the data from sensors, but I couldn't get it to work. Is there a different system for that?

Sorry for all the questions, and thank you in advance!

dlech, 2023-07-26:
1. Reading files in Python is plenty fast if you don't open the file every time you read it. Instead call `seek(0)` on the file before reading again. This applies to any programming language.

2. Yes, the frequency is counts per second which is proportional to rotational velocity.

3. To control motors directly, write `run-direct` to the `command` attribute then write the duty cycle (-100 to +100) to the `duty_cycle_sp` attribute. http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#tacho-motor-subsystem

4. For supported sensors, there is `/sys/class/lego-sensor/`. The most efficient way to get sensor data is using the `bin_data` attribute.  http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#the-lego-sensor-subsytem

JonathanDrukker, 2023-09-09:
Thank you very much!

closed 2023-09-09 by JonathanDrukker

JonathanDrukker, 2024-07-17:
Hello, I am here just to share the code implemented for this in pybricks, and you can use it via EV3DEV by removing the `@micropython.native` decorators in all the function. NOTE: if accidentally one of the cables of the motor unplugs, YOU NEED TO call MotorPorts().