ev3dev/ev3dev #1575: Using MPU6050 on i2c-0 with eeprom

juliovendramini, 2023-07-04:
**Question**

Hi, I want to put a mpu-6050 on the same bus of eeprom to using hardware i2c-0. I alread soldered the giroscope on Tests Points on the lego ev3 board(SCL and SDA), but the eeprom uses basic all the adress on the bus, include 0x68 (id of mpu6050) I alread search a workaround on the internet, but I don't find anything diferent from "fisical disconnect the eeprom from device", do you have an ideias? 

When I use i2cdetect, the output is this:
---------------------------------
robot@ev3dev:~$ sudo i2cdetect -y 0
[sudo] password for robot:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    01 02 03 04 05 06 07 -- 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 -- -- 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
--------------------------------------
robot@ev3dev:~$ uname -a
Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3+ #5 PREEMPT Wed Jun 28 13:28:33 UTC 2023 armv5tejl GNU/Linux
----------------------------------
I alread recompile the kernel with some modifications, but the default kernel show the same thing. I change de DTB to disable eeprom driver and disable bluetoth too, and create the modules for mpu6050.

Sorry about my english. And thanks for the work with EV3dev. A use this since 2018.

dlech, 2023-07-04:
You should not have to disable the eeprom since it uses a different I2C address. Can you share the changes you made to the device tree and kernel configuration?

juliovendramini, 2023-07-04:
Hi, thanks for the fast reply.
After you answer me, I tried to create a new microsd card with lego default mycropython image. 
I2c works correctly. After install the kernel version that I compiled, the problem returns. 
On kernel I only disabled bluetooth suport and add mpu6050 drivers.
I will try to compile the kernel without any modifications and try it.
I will send news after thats.

Thanks.

juliovendramini, 2023-07-05:
Hi, I made it work.
It seen that unselect bluetooth ev3 driver broke i2c-0. When I checked it again, it works. 
Now I have mpu6050 working on hardware i2c-0;

In time, I made a boot modification and change clock to 504mhz and made a overvolt to 1.3V.
It is working pretty good. 250 bogomips

dlech, 2023-07-05:
I am impressed. Did you have to modify the hardware to get the 1.3V? It would be fun to see a picture of how you attached the IMU.

juliovendramini, 2023-07-05:
The result using the mpu6050
[image]
[image]

I also tried to overclock to 550mhz But on boot the SPI bus crash... Maybe if we can slow down the SPI bus...

dlech, 2023-07-05:
Amazing work! Other than replacing the screen, I haven't seen any EV3 hardware hacking like this. Very cool.