ev3dev/ev3dev #1576: Memory swapping

adamszczygielski, 2023-07-11:
Why does ev3dev create a swap upon boot-up? It looks like there is enough available memory, but it allocates swap space. 
I am concerned it may have huge impact on the SD card lifespan.

Here is my case:
```
robot@ev3dev:~$ free -m
              total        used        free      shared  buff/cache   available
Mem:             56          15           9           0          31          36
Swap:            95          10          85
```

dlech, 2023-07-11:
Memory intensive tasks like installing packages using `apt` will fail without additional swap memory. Technically Debian stretch has a minimum system requirement of 128MB of RAM.

dlech, 2023-07-11:
The swap memory is implemented using compressed RAM (zram) and therefore never touches the SD card.