ev3dev/ev3dev #1578: Trying to install pip package inside python-mircopython environment, fell down a rabbit hole

OverclockedToaster, 2023-08-07:
Hi, yesterday my son and I tried to install a pip package (logging) inside the python-micropython env. We both have almost no experience with python envs so we just tried to install it using pip and see what happens. We then noticed that pip wasn't even installed so we tried to update it using `apt`. This didn't work because there was no dns configured on the EV3 but I got `ping google.com` to work after temporarily editing the resolv.conf (apparently, you need to install a service (with apt!) to change it permanently) and connecting the EV3 using a Mini USB cable instead of Bluetooth. However, `sudo apt update` still threw a lot of errors:
```

Ign:1 http://security.debian.org stretch/updates InRelease
Ign:2 http://httpredir.debian.org/debian stretch InRelease
Err:3 http://security.debian.org stretch/updates Release
  404  Not Found
Ign:4 http://httpredir.debian.org/debian stretch Release
Ign:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Hit:10 http://archive.ev3dev.org/debian stretch InRelease
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Err:5 http://httpredir.debian.org/debian stretch/non-free armel Packages
  404  Not Found
Ign:6 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:7 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:8 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:11 http://httpredir.debian.org/debian stretch/main armel Packages
Reading package lists... Done
E: The repository 'http://security.debian.org stretch/updates Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
We then tried to manually get pip on the robot by uploading a pip source file to it with WinSCP the installation script gave us even more errors and after some digging we found out that the standard Python of the OS was 2.7. Our last idea was to try and manually upload a Python 3 tar and compile it from there (as there seems to be no precompiled version available) which ALSO failed because there were some compiler tools missing and we were back at the beginning trying to get apt to work to install the required tools.  
It's sounds a bit complicated, so here's the TL;DR version:
- How do you install a package inside the micropython env?
- How do I get sudo apt update to work?
- How do I install Python 3.11 and pip on the EV3?

Any help would be appreciated

dlech, 2023-08-07:
MicroPython is not Python, so packages from PyPI will not work. (Also, ev3dev has such an old version of Python that `pip` no longer works).

`pybricks-micropython` (v2) ships with [micropython-lib](https://github.com/pybricks/micropython-lib) which includes some subsests of popular modules from the standard library and a few other. So you should be able to `import logging` already.

If you are using Pybricks v3, we haven't added micropython-lib yet. But you can copy the files to your project and use them that way for now.

You can also check upstream for the [latest version of micropython-lib](https://github.com/micropython/micropython-lib).

OverclockedToaster, 2023-08-07:
I am in fact using v3. Which files do you mean exactly and where do I have to copy them? Btw, thanks for the quick response, you guys here are fast🚀

dlech, 2023-08-07:
You can print `sys.path` to see where it looks for files when importing modules (`''` is the current working directory and `.frozen` is a special entry for Pybricks modules, not a real folder).

And you can add any directory you want to this if you want to include a different directory. For example, you could create `/home/robot/lib/` and then put any `.py` files you want there, ones you created or ones you copied from `micropython-lib` or somewhere else. Then at the beginning of your program...

```python
import sys

sys.path.append('/home/robot/lib')

import logging  # imports home/robot/lib/logging.py or /home/robot/lib/logging/__init__.py

...
```

OverclockedToaster, 2023-08-07:
You are my hero, we nearly wasted 3 hours on this! We'll try it out tomorrow directly! Thanks alot

OverclockedToaster, 2023-08-11:
Okay, so we tried it out a few days ago but we unfortunately get the following error:
```
Traceback (most recent call last):
  File "/home/robot/Book/New_test.py", line 9, in <module>
ImportError: no module named 'logging'
```

Here's my code:
```
#!/usr/bin/env pybricks-micropython
import sys
sys.path.append('/home/robot/lib/')
import logging
```

I dropped the folders from https://github.com/micropython/micropython-lib into the `/home/robot/lib` directory:
[image]

I noticed that it works if I move the logging.py file to `/home/robot/lib` but I don't want to move every single file to the lib folder. Is this a problem with sys.path.append not working recursively? But in your example it looks like recursion is working...?

> import logging  # imports home/robot/lib/logging.py or /home/robot/lib/logging/__init__.py

dlech, 2023-08-11:
Personally, I would just copy the modules I was actually going to use instead of copying everything. But if you want to do it this way, then just change the path you add to `sys.path`.  (`sys.path.append('/home/robot/lib/python-stdlib')`)

OverclockedToaster, 2023-08-14:
The first thing you said worked(copying only the needed .py files over), the second method throws an `AttributeError: 'module' object has no attribute 'debug'` when I run this code:
```
#!/usr/bin/env pybricks-micropython
import sys
sys.path.append('/home/robot/lib/python-stdlib')
import logging as log
log.debug('Hello World!')
```
How can I fix this error if I wanted to use the second method you provided?

dlech, 2023-08-14:
On further inspection, it looks like the way the directories are structured, you would need to actually use:

```python
sys.path.append('/home/robot/lib/python-stdlib/logging')
```

OverclockedToaster, 2023-08-14:
Oh, I see, this makes total sense! Thank you for all the quick replies!

closed 2023-09-24 by OverclockedToaster