ev3dev/ev3dev #1579: Connect to a bigger screen?

ItsCrispin, 2023-08-09:
Hello, how would I go about connecting the ev3dev brick to a larger and backlit screen? Can I use the usb port on it? Thanks!

dlech, 2023-08-09:
The USB port is 1.1 only so will probably have limited support for any USB displays. 

You can hack the hardware like this: <https://www.ev3dev.org/news/2015/07/24/Color-LCD-on-the-EV3>. But you won't be able to fit a larger display.

You could also write a program to stream a copy of the display to another device like this: <https://github.com/G33kDude/ev3dev-web-remote>.

BTTRG, 2023-09-16:
> The USB port is 1.1 only so will probably have limited support for any USB displays.

Really? From the LEGO Mindstorms EV3 Developer Kit:
> USB 2.0 Communication, Client interface High speed port (480 MBit/s)
> USB 1.1 Communication, Host interface Full speed port (12 MBit/s)

dlech, 2023-09-17:
It lacks the hardware for proper host mode. But if you force the driver in host mode and use a powered hub it might be able to work.