ev3dev/ev3dev #1580: My ev3 won't start

Xterplot, 2023-08-10:
When I run ev3 with a card with the latest version of ev3dev it freezes on classic boot. And when I run ev3 with ev3 but with the latest version of jessie, it starts loading the libraries but something goes wrong so it crashes there too. I would like my ev3 to be able to run ev3dev. my ev3 has the latest version of lego mindstorms software.

dlech, 2023-08-10:
You could have a bad microSD card or the could be something wrong with the microSD card slot on the EV3.

Xterplot, 2023-08-10:
before i did another project that needed sd card but then i used 2gb sd card and now 16gb so probably not. And I will add that it reads files from the card

Xterplot, 2023-08-10:
2gb card is micro sd and 16 gb is micro sd hc so there is not much difference

Xterplot, 2023-08-11:
I really want to try your ev3dev

dlech, 2023-08-11:
As I said already, either the cards are no good or there is something wrong with the card slot in the EV3 and you need a different EV3.

Xterplot, 2023-08-11:
It wasn't possible to improve.

Xterplot, 2023-08-11:
what sd cards do you use

dlech, 2023-08-11:
I use SanDisk Extreme (be sure to buy from authorized dealer to avoid fakes).

Xterplot, 2023-08-11:
will it work with a usb stick? and one more question, what version of ev3 system will it work with?

dlech, 2023-08-11:
No, it won't work with USB. There is only one EV3 (education and retail are exactly the same).

Xterplot, 2023-08-11:
This is stretch 2020-04-10
https://github.com/ev3dev/ev3dev/assets/141823466/6ae4e462-92ad-463a-b1fe-b71ba76dc724

Xterplot, 2023-08-11:
And jessie 2017 09 14

https://github.com/ev3dev/ev3dev/assets/141823466/2b24eea9-01cc-45ae-88f9-71215719fa32

Xterplot, 2023-08-11:
now what is the problem? can you get drunk somehow?

closed 2023-08-12 by Xterplot