ev3dev/ev3dev #1581: My ev3 with ev3dev won't boot with it.

Xterplot, 2023-08-12:
My ev3 won't start with ev3dev here are videos and photos. 

When i run ev3dev-stretch-ev3-generic-2020-04-10 :

https://github.com/ev3dev/ev3dev/assets/141823466/6ae4e462-92ad-463a-b1fe-b71ba76dc724

When I run ev3dev-jessie-ev3-generic-2017-09-14 :

https://github.com/ev3dev/ev3dev/assets/141823466/2b24eea9-01cc-45ae-88f9-71215719fa32

Can it be repaired somehow?

Xterplot, 2023-08-12:
And in version ev3dev-jessie-2016-10-17 it starts normally.

Xterplot, 2023-08-12:
I wish I could run the latest version

dlech, 2023-08-12:
What LEGO firmware version do have installed on the EV3?

Xterplot, 2023-08-13:
It has Brick HW: V0.60 and Brick FW: V1.09H
https://github.com/ev3dev/ev3dev/assets/141823466/d333f508-8b1c-4fbb-9b2c-a8bb3fd6d143

dlech, 2023-08-13:
You could get an [EV3Console adapter](http://www.mindsensors.com/ev3-and-nxt/40-console-adapter-for-ev3?search_query=EV3Console&results=1) to try to debug. Or you could try a different EV3.

Xterplot, 2023-08-13:
can you do without it?

dlech, 2023-08-13:
we are blind without it

Xterplot, 2023-08-13:
Can I provide some other information instead

Xterplot, 2023-08-14:
Wouldn't it be easier to use the PC input?
https://github.com/ev3dev/ev3dev/assets/141823466/b471a5bb-a68b-4635-82de-201f8437385d

Xterplot, 2023-08-15:
On version ev3dev-jessie-2017-09-14 the thing that stops everything is couldn't mount as ext3 due to feature incompatibilities.
Does what I wrote help?

Xterplot, 2023-08-15:
What version is the most stable?

dlech, 2023-08-15:
https://www.ev3dev.org/downloads/

Xterplot, 2023-08-15:
Maybe make (I'm not saying now) a debug version. That without the ev3 console adapter, logs will be displayed via usb or on the ev3 screen.

Xterplot, 2023-08-18:
I have this adapter and these are the logs
EV3 initialization passed!
Booting EV3 EEprom Boot Loader

         EEprom Version: 0.60
         EV3 Flashtype: N25Q128A13B

EV3 boot system

Jumping to entry point at: 0xC1080000

U-Boot 2009.11 (Oct 26 2012 - 10:30:38)

I2C: ready
DRAM: 64 MB
MMC: davinci: 0
In: series
Out: series
Err: serial
ARM Clock : 300000000Hz
DDR Clock: 132000000 Hz
Invalid MAC address read.
Hit 'l' to stop autoboot: 0
reading boot.scr

864 bytes read
## Executing script at c0600000
reading u-boot.bin

223828 bytes read
## Starting application at 0xC10C0000 ...

U-Boot 2018.05-00309-gebddc0d37d-dirty (May 24 2018 - 23:10:52 -0500)

I2C: ready
DRAM: 64MB
WARNING: Caches not enabled
MMC: davinci: 0
In: series
Out: series
Err: serial
Autoboot in 0 seconds - press 'l' to stop...
553 bytes read in 3 ms (179.7 KiB/s)
Loaded env from uEnv.txt
Importing environment...
Failed to iterate over directory boot
** File not found /boot/boot.scr **
** Unable to read file uImage **
=>

Xterplot, 2023-08-18:
I have this adapter and these are the logs:

https://github.com/ev3dev/ev3dev/files/12378828/logs.txt

dlech, 2023-08-18:
It says that it can't read the files from the microSD card.

```console
** File not found /boot/boot.scr **
** Unable to read file uImage **
```

So as I said originally, it is either a bad microSD card or there is something wrong with the EV3.

Xterplot, 2023-08-19:
Is there anything else i can do instead of buying ev3?

Xterplot, 2023-08-19:
And buy SD cards?

Xterplot, 2023-08-19:
Or maybe you can update from this older version via connect bot by typing sudo ...
What command to enter?

Xterplot, 2023-08-19:
On version https://ev3dev-jessie-ev3-generic-2016-10-17.zip

Xterplot, 2023-08-19:
On this version everything is normal except visual studio code is not working.

closed 2023-08-26 by Xterplot