ev3dev/ev3dev #1582: How to update cmake from 3.7 to 3.13?

f1refa11, 2023-08-17:
**Question**
I wanted to build some app for cmake, but when building using 3.7, the error appears that the minimal required version is 3.13. I tried:
- Adding stretch-backports archive to sources.list (didn't help)
- Building cmake from source(it is SO slow it would at least take a day to build)

Is there a way to simply install cmake 3.13 or higher for ev3dev?

**Desktop (please complete the following information):**
 - OS and version: windows 10 21H2
 - Connection Type: usb
 - Software: ssh

**Robot (please complete the following information):**
 - Device: ev3
 - Versions: latest

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           226.09
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534F237C
BOARD_INFO_TYPE=main
```

dlech, 2023-08-17:
No, I don't think there is a simple way. You would have to build `cmake` from source. You should do this in a docker container instead of on the EV3 so that it isn't so slow.

f1refa11, 2023-08-18:
> No, I don't think there is a simple way. You would have to build `cmake` from source. You should do this in a docker container instead of on the EV3 so that it isn't so slow.

Ok, I'll try it out!

f1refa11, 2023-08-22:
How do I configure CMake to build using ARM toolchain?

dlech, 2023-08-22:
By using a cmake toolchain file.

The `ev3dev/debian-stretch-cross` docker image contains a `/home/compiler/toolchain-armel.cmake` file for this purpose. https://github.com/ev3dev/docker-cross/blob/master/ev3dev-stretch/toolchain-armel.cmake

Then you just set `-DCMAKE_TOOLCHAIN_FILE=...` when configuring your project.

Nikolay-Zagrebin, 2024-06-19:
> I wanted to build some app for cmake, but when building using 3.7, the error appears that the minimal required version is 3.13. I tried:

There is a way, but it is advisable to use a docker image.
The general algorithm, for example for 3.22.2, is as follows

sudo apt update
sudo apt upgrade
sudo apt install build-essential checkinstall zlib1g-dev libssl-dev

wget https://github.com/Kitware/CMake/releases/download/v3.22.2/cmake-3.22.2.tar.gz
tar -zxvf cmake-3.22.2.tar.gz
cd cmake-3.22.2
sudo ./bootstrap
sudo make
sudo make install

cmake --version