ev3dev/ev3dev #1584: "sudo apt-get update" is getting some 404s and "sudo apt-get install build-essential" also fails

TKobe28, 2023-08-20:
I was following this [tutorial](https://www.ev3dev.org/docs/tutorials/getting-started-with-c/).

I need to execute these 2 commands:

- `sudo apt-get update` which outputed this:
`Ign:1 http://security.debian.org stretch/updates InRelease                                       
Err:2 http://security.debian.org stretch/updates Release                                         
  404  Not Found
Ign:3 http://httpredir.debian.org/debian stretch InRelease                
Ign:4 http://httpredir.debian.org/debian stretch Release                  
Ign:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Hit:8 http://archive.ev3dev.org/debian stretch InRelease
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Ign:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Err:5 http://httpredir.debian.org/debian stretch/contrib armel Packages
  404  Not Found
Ign:6 http://httpredir.debian.org/debian stretch/non-free armel Packages
Ign:7 http://httpredir.debian.org/debian stretch/non-free all Packages
Ign:9 http://httpredir.debian.org/debian stretch/main all Packages
Ign:10 http://httpredir.debian.org/debian stretch/main armel Packages
Ign:11 http://httpredir.debian.org/debian stretch/contrib all Packages
Reading package lists... Done
E: The repository 'http://security.debian.org stretch/updates Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: The repository 'http://httpredir.debian.org/debian stretch Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

- `sudo apt-get install build-essential` which outputed this:
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package build-essential is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'build-essential' has no installation candidate`

I have succesfully set up internet connection through usb. I checked through `ifconfig `and `ping google.com`.
Am I missing something?

dlech, 2023-08-20:
Stretch is so old now, the URL for updates no longer works. There is still fix though. Change `/etc/apt/sources.list` to match this: https://github.com/ev3dev/docker-library/blob/master/ev3dev-stretch/layers/debian/etc/apt/sources.list

TKobe28, 2023-08-20:
Thank you very much for amazingly fast response! 

It fixed both issues.

closed 2023-08-20 by TKobe28