ev3dev/ev3dev #1585: Uncompressed ev3dev iso.

Xterplot, 2023-08-26:
I need uncompressed ev3dev version:
ev3dev-stretch-ev3-generic-2020-04-10.img

Would it be possible to download such a zip or folder?

dlech, 2023-08-26:
You can flash it to an SD card or USB drive and look at the files that way. You will need a way to mount the Linux ext4 file system, so it won't work on Windows or Mac (without special software anyway).

Xterplot, 2023-08-26:
Do you remember my previous issues so i fixed one thing but it still doesn't work:
https://github.com/ev3dev/ev3dev/assets/141823466/2b24eea9-01cc-45ae-88f9-71215719fa32

I saw some post how to fix it but i needed to do hex for it but i was uncompressing in 7z it didn't work some files and so ev3dev might be corrupted. And if I do it, it says the same as on ev3, i.e. something with ext-3 and ext-4.

And can you help me with this?

dlech, 2023-08-26:
I'm not sure what that link is. The official download is https://github.com/ev3dev/ev3dev/releases/download/ev3dev-stretch-2020-04-10/ev3dev-stretch-ev3-generic-2020-04-10.zip.

Xterplot, 2023-08-26:
This is a link to a photo of what happened.
But if you want to write it here

EXT4-fs (mmcblk0p2): couldn't mount 5 Ext due to feature incompatibilities

https://github.com/ev3dev/ev3dev/assets/141823466/b22247c0-f265-4f18-ab65-27972f3563c9

ore info in the picture.

closed 2025-02-17 by Arcodyq