ev3dev/ev3dev #1586: Command Line on EV3 Brick without using SSH

BigKitten, 2023-08-27:
The EV3 default firmware has a nice on screen keyboard to do thing such as rename files.
It would be nice to see a similar implementation along with a new terminal tab to allow you to type and run command directly on the EV3 brick without SSH.

Xterplot, 2023-08-27:
Why can't you use ssh? Don't have an internet adapter? Try this:
 https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth

And then you can use a tool like connect bot (for phones) or putty (for pc)

dlech, 2023-08-27:
You can do this already. Just attach a USB keyboard to the EV3 and press <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F6</kbd> to switch to a login prompt. Press <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F1</kbd> to switch back to the Brickman interface.

BigKitten, 2023-08-27:
Sorry if I was unclear, but what I meant that it would be nice to do this without even a keyboard, and using an on-screen keyboard like the one in the default lego-supplied firmware.

BigKitten, 2023-08-30:
@dlech i just found the password input keyboard while connecting to Wi-Fi, it would be cool to have a builtin terminal using that on-screen keyboard.

BTTRG, 2023-10-08:
I think that would make command entry slow and cumbersome. Using the framebuffer terminal with a USB keyboard is probably best. My only problem with that terminal is that the cursor isn't visible by default, however you can enable it upon login by amending your `.bashrc`:

```bash
# enable cursor by default
echo -e '\033[?25h' > $(tty)
```

pythoncat1, 2023-10-08:
I agree, although it would still be nice to have a way to accept text inputs right on the brick from, for example, a python program.