ev3dev/ev3dev #1588: Overcloacking EV3

JonathanDrukker, 2023-09-09:
Hey,
In the uEnv file on the microSD of the EV3 is written the following:
```
# uncomment to increase CPU frequency
# 375MHz is rated speed at available voltage (1.2V)
# 456MHz is max rated speed of CPU (supposed to have 1.3V for this)
# cpufreq=375
```

My question is, if I uncomment `cpufreq=375` and change the value to 456, what will the effects be?
Will it make the CPU cycle time quicker?
Will nothing change?

Thank you in advance,
Jonathan

dlech, 2023-09-09:
Yes, it will run a little faster. I wrote a blog post about it. https://lechnology.com/2018/06/overclocking-lego-mindstorms-ev3-part-2/

JonathanDrukker, 2023-09-10:
Thank you very much!

closed 2023-09-10 by JonathanDrukker