ev3dev/ev3dev #1589: Turn off the audible terminal bell

BTTRG, 2023-09-21:
Every time I press an incorrect key or get to the end of a `less` screen while in a terminal on the EV3 itself, I get blasted with an annoying and obnoxious beep from the EV3 speaker. I've tried all manners of solutions and I could never disable it. There isn't even a `pcspkr` module you can cripple.

Also, when using `modprobe` you must prefix it with with `/sbin/` in order for it to work. `kmod` and `lsmod` don't require this, but everything else does. I think the PATH is messed up.

dlech, 2023-09-22:
This Q&A looks like it might have some useful suggestions: https://unix.stackexchange.com/q/73672/91467

BTTRG, 2023-09-23:
Well, I read through that page again and turns out I had `set bell-style none` commented out the whole time. However, this doesn't disable the beep in `less` or `nano`. I tried `less -Q` to make it silent but instead it inverts the whole screen in the place of a beep (I assume this is supposed to be a "visible" bell, and you can't un-invert the screen AFAIK). 
Whatever, I can certainly live with the beep gone just for the shell!

closed 2023-09-23 by BTTRG