ev3dev/ev3dev #1591: "apt-get update" completely messes everything up

guobdd-03, 2023-09-26:
Greetings!

**What are you trying to do?**

I am trying to do apt update via SSH, I have already connected to internet via Wi-Fi using TP-LINK dongle (exact model is, unfortunately, no longer known to me).

**What did you expect to happen**

I needed to update apt-get, because otherwise I couldn't install any packages. I expected to update apt-get without any problem, and subsequently start installing other packages.

**What actually happened**

I have typed sudo apt-get update. Update went smoothly but after rebooting, it appeared to completely overwrite the information on SD card, and instead of ev3 launching brickman, it launched standart ev3 OS. So to launch ev3dev software I had to flash a sd card again.

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Connection Type: Wi-Fi
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: below

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535DFF16
BOARD_INFO_TYPE=main

In the end, I want to apologise for my English, and I hope you understand what i meant. Thank you in advance for your sopport.

dlech, 2023-09-26:
If `apt-get update` broke things, then you probably have a bad SD card. If it is larger than 32GB, then that is probably the issue.

guobdd-03, 2023-09-26:
My microSD card is 4gb from an unknown manufacturer. I will get that one microSD card tomorrow at my local shop: https://www.eldorado.ru/cat/detail/karta-pamyati-sdhc-micro-sandisk-microsdhc-ultra-plus-32gb-uhs-i-sdsqub3-032g-gn6ma/

Will report after test, thanks for you quick support!

guobdd-03, 2023-09-27:
Looks like that problem has gone away. I may reopen this issue again if the problem will reappear. As always, thanks for pointing out the issue!