ev3dev/ev3dev #1593: Migration to 32GB Micro SD isn't working

BTTRG, 2023-09-29:
I used `dd` on my iMac to make a verbatim copy of my 4GB ev3dev SD card onto a 32GB one (to expand available storage space). The command I used was:

`sudo dd if=/dev/disk2 of=/dev/disk3`

I booted the 32GB card and systemd starts loading services but it fails to start the full system. It gets stuck on something being timed out. 
This is the 32GB card I used: https://www.amazon.com/SanDisk-Ultra-microSDXC-UHS-I-Memory/dp/B09LTCKWZG/

dlech, 2023-09-30:
The vendor in the link is not an authorized SanDisk reseller, so it is possible you got a knock-off card that is defective.

BTTRG, 2023-09-30:
I didn't get that 32GB Micro SD card from Amazon, I actually bought it in Target so AFAIK it's a genuine SanDisk card. Observing the EV3 screen the boot gets stuck on:

`[ TIME ] Timed out waiting..c.1.auto.device.`

dlech, 2023-09-30:
I would just try flashing the card again using etcher.

BTTRG, 2023-10-03:
Tried Etcher, it works now. I think the problem might've been extFS for Mac automatically "repairing" the disk right after `dd` finishes. Etcher unmounts the target disk when completed.

closed 2023-10-03 by BTTRG