ev3dev/ev3dev #1594: Installing NodeJS on ev3dev

Madelyn-of-Hell, 2023-10-01:
**What are you trying to do?**

I am trying to install Node.js and NPM onto my ev3 so that I can run Javascript programs through the use of WasabiFan's ev3dev package

**What did you expect to happen**

The command (sudo apt-get install npm nodejs, sourced through another github page on the subject) should've downloaded node.js onto the ev3.

**What actually happened**

The command threw the error

Package nodejs is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'nodejs' has no installation candidate

**Source Code**

Here is the program I am trying to run.

```
sudo apt-get install nodejs
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: macOS 13
 - Connection Type: USB
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=001653815515
BOARD_INFO_TYPE=main
```

**Background Information**

I am trying to use Javascript because I want to understand how it compares to python in processing speeds, input time, precision, and functionality.

dlech, 2023-10-01:
Did you do `sudo apt update` first?

Madelyn-of-Hell, 2023-10-02:
I altered the file as instructed, `yet running sudo` apt update returned the same error message:

```
Err:1 http://archive.ev3dev.org/debian stretch InRelease                
  Could not resolve 'archive.ev3dev.org'
Err:2 http://archive.debian.org/debian stretch InRelease                
  Could not resolve 'archive.debian.org'
Reading package lists... Done        
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://archive.debian.org/debian/dists/stretch/InRelease  Could not resolve 'archive.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

dlech, 2023-10-02:
Is the EV3 connected to the internet?

Madelyn-of-Hell, 2023-10-03:
I assumed that its connection to my computer allowed it to access the internet. I've tried activating wifi on the ev3 alone, but it says that it's not available. how do I connect it directly?

dlech, 2023-10-03:
Please see https://www.ev3dev.org/docs/networking/ for the options.