ev3dev/ev3dev #1595: NXT TouchSensor not detected on EV3

jenny1research, 2023-10-02:
**What are you trying to do?**

I have been using MicroPython on Visual Studio Code to program Lego Mindstorms EV3. 

I am trying to write a program to detect my touch sensor. While my brick is EV3, the touch sensor is from the NXT kit.

**What did you expect to happen**

For the EV3 brick to detect the TouchSensor.

**What actually happened**

The error I get is: 
Starting: brickrun --directory="/home/robot/my_project" "/home/robot/my_project/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/my_project/main.py", line 23, in <module>
OSError: [Errno 19] ENODEV: 

A sensor or motor is not connected to the specified port:
--> Check the cables to each motor and sensor.
--> Check the port settings in your script.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.

----------
Exited with error code 1.

**Source Code**

Here is the program I am trying to run.

```
# Create your objects here. 

ev3 = EV3Brick() 

left_wheel = Motor(Port.B)  

right_wheel = Motor(Port.C)  

robot = DriveBase(left_wheel,right_wheel,wheel_diameter=56,axle_track=100)  

touch = TouchSensor(Port.S1) 

# Write your program here. 

ev3.speaker.beep() 

while True: 

if touch.pressed() == True: 

robot.turn(100) 

```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: macOS 10.13.6
 - Connection Type: USB
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: SD card

jenny1research, 2023-10-02:
Thank you for your prompt response. 

I have revised my code as shown below, but I am still having issues. I removed the TouchSensor from the line "from ev3devices import ..." and instead included "from nxtdevices import TouchSensor".

Could you please let me know how else I should revise this?

```
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

from nxtdevices import TouchSensor

# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()
left_wheel = Motor(Port.B) 
right_wheel = Motor(Port.C) 
robot = DriveBase(left_wheel,right_wheel,wheel_diameter=56,axle_track=100) 

touch = TouchSensor(Port.S1)

# Write your program here.

while True: 
    if touch.pressed() == True: 
        robot.turn(100)

```

This is the error message that I am getting.

Starting: brickrun --directory="/home/robot/my_project" "/home/robot/my_project/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/my_project/main.py", line 10, in <module>
ImportError: no module named 'nxtdevices'
----------
Exited with error code 1.

dlech, 2023-10-02:
Oops, I missed adding `pybricks.` in my suggestions. It should be:

```python
from pybricks.nxtdevices import TouchSensor
```

jenny1research, 2023-10-02:
Thank you for your help, the NXT TouchSensor is now being detected! Would the same modification apply to NXT sound sensor and NXT ultrasonic sensor?

```
from pybricks.nxtdevices import (TouchSensor, SoundSensor, UltrasonicSensor)
```

dlech, 2023-10-02:
> Would the same modification apply to NXT sound sensor and NXT ultrasonic sensor?

Yes, it applies for all NXT sensors, so those and more, e.g. `ColorSensor`.

jenny1research, 2023-10-02:
> #!/usr/bin/env pybricks-micropython
> from pybricks.hubs import EV3Brick
> from pybricks.ev3devices import (Motor, ColorSensor,
>                                  InfraredSensor, UltrasonicSensor, [...]

Thank you for the clarification. It is much appreciated.

I am now trying to create a loop to react to the touch sensor. However, the below code makes the EV3 turn but it does not go straight when the touch sensor is pressed. When the first three lines of coding is written without the else statement, the touch sensor works as intended. Could you let me know which part should be modified?

```
# Write your program here.
while True: 
    if touch.pressed() == True: 
        robot.straight(100)
    else:
        robot.turn(10000)
```

dlech, 2023-10-03:
It depends on what the desired behavior of the robot is. But I'm guessing you want something like 

```python
# Write your program here.

speed, _, turn_rate, _ = robot.settings()

while True: 
    if touch.pressed() == True: 
        robot.straight(100)
    else:
        robot.drive(speed, turn_rate)
        wait(10)
```

`DriveBase.drive()` is not a blocking function, so it doesn't take any time. It just starts motors or updates the speed if already running then the program continues immediately.

On EV3, it is also important to always have some sort of delay in long running loop like to allow time for the Linux OS to keep running in the background. In this case, I have added a `wait()` for this.

jenny1research, 2023-10-05:
> ```python
> # Write your program here.
> 
> speed, _, turn_rate, _ = robot.settings()
> 
> while True: 
>     if touch.pressed() == True: 
>         robot.straight(100)
>     else:
>         robot.drive(speed, turn_rate)
>         wait(10)
> ```

Thank you for all your help. 

With the above program, the robot goes straight when the touch sensor is pressed, but it does not move otherwise. 

I am now hoping to program the robot to go straight when the touch sensor is not pressed, and have the robot turn when the touch sensor is pressed. Would this be possible?

dlech, 2023-10-05:
I would `print(speed, turn_rate)` to see what the values are. Maybe speed is 0 which is why it isn't moving?

To rephrase the desired behavior, it sounds like you want the robot to go straight until the the touch sensor is pressed, then turn, then repeat.

So this could be written as:

```python

while True:
    # start driving straight and continue with the program
    robot.drive(500, 0)

    # wait until touch sensor is pressed
    while not touch.pressed():
        wait(10)

    # turn 90 degrees
    robot.turn(90)

    # then loop back and start driving straight again
```

dlech, 2023-10-10:
What is line 18 in your program?

To reboot, just turn the EV3 off and turn it back on again.

dlech, 2023-10-11:
Are you using medium or large motors? If you are using medium, then that would explain going straight instead of turning. To fix, you would need to change the positive direction on one of the motors.