ev3dev/ev3dev #1596: Brick not boting to ev3dev

nezmar48, 2023-10-04:
**What are you trying to do?**

I am trying to boot the brick to ev3dev.

**What did you expect to happen**

It should have started the ev3dev boot splash and start the ev3dev OS.

**What actually happened**

Instead, it starts the normal ev3 booting sequence and boots to the normal OS.

I am using an EV3 with the 1.10E firmware, a 4GB SDHC flashed using balenaEtcher and ev3dev-stretch-ev3-generic-2020-04-10.img file. 
The SD card is recognised by the EV3 when formatted to default settings. 
Everything is unplugged from the brick ecxept the cable charging the removable battery that is fully charged.

I have tried reflashing it but it didn't work. Not even after trying out the official ev3micropython version.

dlech, 2023-10-04:
Sometimes the SD card doesn't make good connection in the SD card slot.

dlech, 2023-10-06:
Maybe you accidentally "repaired" the SD card after flashing it with ev3dev? Windows and Mac don't like the disk format and will ask you to "repair" it which actually breaks it.

BTTRG, 2023-10-08:
Try removing the SD card, booting to stock EV3 firmware, shutting down from there, then insert the SD card and see if ev3dev will boot. If you have Paragon extFS installed, disable it from accessing the SD card and try flashing again.

Xterplot, 2023-10-11:
I had a similar problem, use rufus: https://rufus.ie

This is an application that is used to flash SD cards and other media with ISO files. others recommend this application for flashing with Linux and when I did it with Baltea Etcher it didn't work and I had a similar problem but when I tried with Rufus it worked.

nezmar48, 2023-10-11:
@Xterplot Thanks this solved the issue.

closed 2023-10-11 by nezmar48