ev3dev/ev3dev #1599: Network error: Connection timed out!

pichenka007, 2023-11-13:
I'm trying to connect to ev3 via ssh

he should connect

It seems to connect via bluetooth but doesn’t want to connect via ssh from which device I wouldn’t try

**Desktop:**
 - OS and version: Window 7, android 10
 - Connection Type: Bluetooth
 - Software: PuTTY, ConnectBot

**Robot:**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10.img

```
I can't connect ssh lol
```

Actually, I'm making a 2D printer for a competition, but damn, how do I connect this damn ssh

I really hope for more attention

dlech, 2023-11-13:
Can you give the exact steps to reproduce the problem? All of the clicks, button presses, etc. Screen shots if possible.

pichenka007, 2023-11-14:
I don't know. For example, I was just able to connect to it using a USB modem from my phone (ConnectBot)
and already from the PC via PuTTY I have a "timeout error"

pichenka007, 2023-11-14:
here is my telegram:
@pichenka007gd

pichenka007, 2023-11-14:
I'll try it now via Xshell

pichenka007, 2023-11-14:
no, it's the same with Xshell

pichenka007, 2023-11-14:
It connects through a tablet that distributes the Internet, but not through a PC.

pichenka007, 2023-11-14:
[image]
[image]
[image]
[image]
[image]

pichenka007, 2023-11-14:
I’ll explain in the 1st picture I enter the ip
in the second picture I turn on the checkbox for connection attempts (this is under the 1st enable checkbox)
on the rest it is written

connecting to ssh

The authenticity of the server cannot be established

server key fingerprint *****-*******************

want to continue trying to connect (yes)

key exchange algorithm: ******************

we use the algorithm: **********************************

trying to log in

attempt to log in using public keys from memory

trying to log in

attempt at "interactive keyboard" authorization

then I enter the password (maker)

and everything works

pichenka007, 2023-11-14:
))) ssh 192.168.42.146
Connecting to 192.168.42.146:22...
Could not connect to '192.168.42.146' (port 22): Connection failed.
this is in Xshell

pichenka007, 2023-11-14:
[image]

pichenka007, 2023-11-14:
[image]

pichenka007, 2023-11-14:
[image]

pichenka007, 2023-11-14:
[image]

pichenka007, 2023-11-14:
Well, this is in PuTTY

pichenka007, 2023-11-14:
uh but that's bad, now in bluetooth and wifi it says "Not avaible

closed 2023-11-19 by pichenka007