ev3dev/ev3dev #1600: Type hints not working

bitbaggi, 2023-11-14:
Type hints not working in ev3dev, is that right?

something like this not working.

```
#!/usr/bin/env python3

from threading import Thread
from ev3dev2.motor import OUTPUT_A
from ev3dev2.motor import OUTPUT_D
from ev3dev2.motor import LargeMotor, SpeedPercent

large_motor_a: LargeMotor = LargeMotor(OUTPUT_A)
large_motor_d = LargeMotor(OUTPUT_D)

thread_a = Thread(target=large_motor_a.on_for_seconds, args=(SpeedPercent(100), 4, True))
thread_b = Thread(target=large_motor_d.on_for_seconds, args=(SpeedPercent(100), 4, True))

thread_a.start()
thread_b.start()

thread_a.join()
thread_b.join()
```

dlech, 2023-11-14:
You need to create a virtual environment and install the `ev3dev2` package and tell your IDE to use the virtual environment. Then it should have code completion.

bitbaggi, 2023-11-14:
I have installed it and code completion is working.

But when i load the python script to the ev3 it raises an exception that the symbol : is not expected.

I can add an photo this evening.

dlech, 2023-11-14:
The title of the issue says that type hints aren't working, so I assumed that was the problem.

On the line
```python
large_motor_a: LargeMotor = LargeMotor(OUTPUT_A)
```

You need to change `:` to `=`.

dlech, 2023-11-14:
Oh, right, at runtime. ev3dev-stretch has Python 3.5 which doesn't support type hints, so you won't be able to use them.