ev3dev/ev3dev #1603: Aw snap, my EV3 LCD isn't working!

luRaichu, 2023-12-01:
Today I cleaned my EV3's LCD with compressed air (cause it was full of dust) and after re-plugging the FPC cable (and trying again multiple times) the screen wouldn't turn on. Then, to top it all off I accidentally put a minor crack in the bottom-right corner of the LCD. I guess I need a new one!
[This tutorial](https://youtu.be/gPNJC5Uz9HY) requires surface-mount soldering an FPC socket to a breakout board. I can't do surface-mount with pins that tiny. Is there a way to replace the LCD with only through-hole soldering?

luRaichu, 2023-12-31:
I already ordered an FPC cable, breakout and color screen. I'm currently waiting on the breakout but would it be possible to dim/turn off the backlight on the new LCD?

dlech, 2024-01-01:
The devicetree just has it configured as a gpio backlight, so it is just on/off.

luRaichu, 2024-01-03:
Just tried to hook up the Fermion 1.8" 128x160 TFT I got tonight, I'm getting no sign of life. I'll try again tomorrow.
Is there anything I can measure to make sure the EV3 is outputting proper voltages to the FPC connector?

luRaichu, 2024-01-03:
[image]
I redid the wiring but the screen is still blank. And yes, I checked the continuity. Everything's connected and kosher.
I might have possibly killed my TFT in the process of removing unnecessary interfaces.

dlech, 2024-01-03:
It could also just be that the backlight is turned off. So the first thing I would measure is the voltage on the backlight pin. That should be independent of the rest of the display. If you have voltage there and on the VCC pin and the display isn't lighting up, then yeah, you might have broken something.

If there isn't voltage on the backlight pin, then I would SSH into the EV3 to make sure the correct device tree is being used.

luRaichu, 2024-01-03:
There's 3.3v on the backlight pin. The screen must be dead.

luRaichu, 2024-01-09:
I just received a brand new replacement TFT, I wired it up in the same fashion but it still won't light up. There's 3.3v on the backlight pin, but for a moment I accidentally wired CSDIS (EV3) -> M_CS_SD (TFT) and MISODIS  (EV3)-> BLK (TFT). I powered on the EV3 with the screen attached but then shut it off after I noticed this error, corrected it, and it still won't light up. Did the wiring error break something?

dlech, 2024-01-09:
Is there voltage on the VCC pin? Are you sure the EV3 is using the correct device tree file?

luRaichu, 2024-01-09:
Measuring VCC against GND, there's only half a volt flowing through. As for the device tree I'm not sure how to evaluate and correct it.

dlech, 2024-01-09:
You should be able to SSH into the EV3 and get a terminal so you can inspect the kernel logs/sysfs to see what is loaded.

But VCC should be 3.3V and be always on, so maybe there is a problem with the wiring?

luRaichu, 2024-01-10:
Turns out the wiring was backwards. When using the Tinkersphere FPC breakout for this project you must follow the pin numbering from the opposite side.
Now, the TFT lights up but then switches to a static pattern. Am I supposed to enable the Adafruit driver in uEnv.txt?

dlech, 2024-01-10:
> Am I supposed to enable the Adafruit driver in uEnv.txt?

yes

```
# uncomment this line if you have an Adafruit 1.8" screen
fdtfile=da850-lego-ev3-adafruit18.dtb
```

luRaichu, 2024-01-10:
Alright, it's finally working now, however for my mod I want the screen rotated 180 degrees from what it is now. I'll also try to compile the ev3dev kernel and correct the 1 column and 2 rows that are glitched out.
[image]

dlech, 2024-01-10:
Finally! That's really great!

FYI, you can use the `da850-lego-ev3-color.dtb` instead of the adafruit18 one and it should take care of the rotation. But yeah, will probably have to modify the kernel driver to take care of the bad pixels on the edges.

luRaichu, 2024-01-10:
Ooh, I'm booting right now with the `da850-lego-ev3-color.dtb` device driver and the bad pixels are gone...

luRaichu, 2024-01-10:
[image]
With `da850-lego-ev3-color.dtb` the screen is still offset though. What do I add to the st7735r.c driver to correct it?

luRaichu, 2024-01-10:
I'm attempting to compile the ev3dev-stretch kernel on Xubuntu 23.10 (even though I haven't updated the st7735r driver yet) and I'm hit with this make error:
```
make[1]: Leaving directory '/home/bigboys/kernelbuild/ev3/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
make: *** [Makefile:146: sub-make] Error 2
make: Leaving directory '/home/bigboys/kernelbuild/ev3/ev3-kernel'
```
I'm following [this tutorial](https://github.com/ev3dev/ev3dev-buildscripts/blob/ev3dev-stretch/README.md).

luRaichu, 2024-01-10:
Here's the full build log in all its glory:
https://pastebin.com/3EpqyzHx

dlech, 2024-01-10:
It sounds like the wrong assembler program is being used (one that doesn't support GCC extensions).

luRaichu, 2024-01-10:
I successfully compiled the x86 Linux kernel (6.7) earlier...

dlech, 2024-01-10:
Maybe `ARCH=arm` or `CROSS_COMPILE=arm-linux-gnueabi-` is missing? Or there is some leftover generated file from the previous build interfering?

dlech, 2024-01-10:
> What do I add to the st7735r.c driver to correct it?

I thinks something like this: https://github.com/ev3dev/ev3-kernel/blob/ad1f6c8727bce29ec400c9385f7f72ee2c0e59fc/drivers/gpu/drm/tinydrm/st7586.c#L221

luRaichu, 2024-01-11:
How do I turn off the backlight?

dlech, 2024-01-11:
One way it to poke it directly via `/sys/class/backlight/`.

luRaichu, 2024-01-11:
Backlight brightness lives at `/sys/class/backlight/backlight/brightness`, but you can only turn it on/off if you have the backlight pin connected. You can switch it like this:
```
echo 0 | sudo tee /sys/class/backlight/backlight/brightness
```
I'd like to fix it so you can do this without root. And perhaps you could simply use PWM for dimming?
As for those pesky glitched pixels, I built a custom kernel on a Ubuntu LTS VPS with this mod to the st7586 C driver like you said:
```
mipi_dbi_command(mipi, MIPI_DCS_SET_PARTIAL_AREA, 0x01, 0x02, 0x03, 0x77);
```
This has no effect, probably because the Fermion and Adafruit TFTs actually use the st7735r driver instead. However, there are some clues in `/include/video/mipi_display.h`:
```h
/* MIPI DSI Processor-to-Peripheral transaction types */
enum {
	MIPI_DSI_V_SYNC_START				= 0x01,
	MIPI_DSI_V_SYNC_END				= 0x11,
	MIPI_DSI_H_SYNC_START				= 0x21,
	MIPI_DSI_H_SYNC_END				= 0x31,
```
I'll play with these and report if they do anything.

luRaichu, 2024-01-11:
> ```c
> /* MIPI DSI Processor-to-Peripheral transaction types */
> enum {
> 	MIPI_DSI_V_SYNC_START				= 0x01,
> 	MIPI_DSI_V_SYNC_END				= 0x11,
> 	MIPI_DSI_H_SYNC_START				= 0x21,
> 	MIPI_DSI_H_SYNC_END				= 0x31,
> ```
> 
> I'll play with these and report if they do anything.

These do nothing.

dlech, 2024-01-11:
> And perhaps you could simply use PWM for dimming?

Possible, yes. Simple, no. 

The pin it is connected to doesn't have a hardware PWM that can be muxed. There is a bitbang gpio-pwm driver, but that would cause the screen to flicker because it isn't a hardware PWM with perfect timing. So the only way to make it work would be add another PWM to the PRU firmware like we did for the status lights.

> This has no effect, probably because the Fermion and Adafruit TFTs actually use the st7735r driver instead.

I meant that the line should be copied and pasted to the st7735r driver. It uses (mostly) the same command set. There may be an additional command required to enable it too (`MIPI_DCS_ENTER_PARTIAL_MODE`?). The datasheet should hopefully explain this (note: the datasheet calls command 0x30 `PTLAR` but is the same value as `MIPI_DCS_SET_PARTIAL_AREA`)

dlech, 2024-01-11:
> I'd like to fix it so you can do this without root.

Normally, the display manager would do this. But our display manager (brickman) was designed mainly for the default display that doesn't have a backlight so we never included this.

Perhaps the simplest thing to do would be to add a udev rule similar to `/lib/udev/rules.d/60-ev3dev.rules` to allow users in the `ev3dev` group access to the file in sysfs. (Don't modify this file, create a new on in `/etc/udev/rules.d/`.)

Or you could make a daemon program that runs in the background  as root and uses some sort of IPC to allow users control the backlight.

luRaichu, 2024-01-12:
> Perhaps the simplest thing to do would be to add a udev rule similar to `/lib/udev/rules.d/60-ev3dev.rules` to allow users in the `ev3dev` group access to the file in sysfs. (Don't modify this file, create a new on in `/etc/udev/rules.d/`.) Pro tip: don't forget the newline at the end of the file - IIRC this version of udev will ignore the last line of the file without it.
> 
> ```
> SUBSYSTEM=="backlight", ACTION=="add", \
>     RUN+="/usr/bin/find /sys$devpath -type f -exec /bin/chmod g+u {} + -exec /bin/chown :ev3dev {} +"
> ```

That works. As for the st7735r driver, I added
```
mipi_dbi_command(mipi, MIPI_DCS_SET_PARTIAL_AREA, 0x00, 0x00, 0x00, 0x77);
```
Then I changed `MIPI_DCS_ENTER_NORMAL_MODE` to `MIPI_DCS_ENTER_PARTIAL_MODE` and this is the result
[image]
There are still bad pixels on the bottom of the screen (which are not visible in the photo). I don't think the `MIPI_DCS_ENTER_PARTIAL_MODE` command can change the actual offset of the display?

dlech, 2024-01-12:
Maybe that is the wrong command. Looking deeper, think we need to apply this patch series. https://patchwork.freedesktop.org/series/71567/ Then there will be a way to add the required offsets needed.

luRaichu, 2024-01-12:
Yeah, the ev3dev kernel is too old for this patch to directly apply. You'd have to backport it manually, which is beyond my expertise.

dlech, 2024-01-12:
I think the key patch is this one. https://patchwork.freedesktop.org/patch/348843/?series=71567&rev=3

You could search for `MIPI_DCS_SET_COLUMN_ADDRESS` and `MIPI_DCS_SET_PAGE_ADDRESS` and add the offsets there, just hard-coded instead of adding the `left_offset` and `top_offset` to the struct.

luRaichu, 2024-01-13:
Is there any documentation on the standard use/parameters of these MIPI commands? That would help me figure out a couple things on my own.

luRaichu, 2024-01-13:
The `SET_COLUMN` and `SET_PAGE` commands live in `drm_mipi_dsi.c`, but I'm not sure what to modify here.
```c
/**
 * mipi_dsi_dcs_set_column_address() - define the column extent of the frame
 *    memory accessed by the host processor
 * @dsi: DSI peripheral device
 * @start: first column of frame memory
 * @end: last column of frame memory
 *
 * Return: 0 on success or a negative error code on failure.
 */
int mipi_dsi_dcs_set_column_address(struct mipi_dsi_device *dsi, u16 start,
                                    u16 end)
{
        u8 payload[4] = { start >> 8, start & 0xff, end >> 8, end & 0xff };
        ssize_t err;

        err = mipi_dsi_dcs_write(dsi, MIPI_DCS_SET_COLUMN_ADDRESS, payload,
                                 sizeof(payload));
        if (err < 0)
                return err;

        return 0;
}
EXPORT_SYMBOL(mipi_dsi_dcs_set_column_address);
```
I don't really use C often.

dlech, 2024-01-14:
> Is there any documentation on the standard use/parameters of these MIPI commands?

The [datasheet](https://cdn-shop.adafruit.com/datasheets/ST7735R_V0.2.pdf) should explain the commands (just look at the numbers since they don't have the same name).

> but I'm not sure what to modify here.

Hopefully you just need to add a few pixels to `start` and `end`. So `start += 2;` and `end += 2` at the beginning of the function.

luRaichu, 2024-01-15:
I did
```c
int mipi_dsi_dcs_set_column_address(struct mipi_dsi_device *dsi, u16 start += 1,
                                    u16 end -= 1)
...
```
But there's no effect.

dlech, 2024-01-15:
You have to add it in the code, not in the function signature. I would expect your change to cause a compile error.

```c
int mipi_dsi_dcs_set_column_address(struct mipi_dsi_device *dsi, u16 start,
                                    u16 end)
{
        const u16 offset = 2; /* change this number as needed */
        start += offset;
        end += offset;
        u8 payload[4] = { start >> 8, start & 0xff, end >> 8, end & 0xff };
        ssize_t err;

        ...
}
```

luRaichu, 2024-01-15:
> You have to add it in the code, not in the function signature. I would expect your change to cause a compile error.
>
> ```c
> int mipi_dsi_dcs_set_column_address(struct mipi_dsi_device *dsi, u16 [...]

I did this instead but still no effect... Pretty sure I'm running the kernel with this change.

dlech, 2024-01-15:
If the first attempt didn't cause a compile error, then I don't think this file is actually being compiled and used.

luRaichu, 2024-01-19:
What do we do next?

dlech, 2024-01-19:
I think this is not the right place to make the change in our version of the kernel.  I think it needs to be changed here:

https://github.com/ev3dev/ev3-kernel/blob/b8e704fe9ed45fcc34a6e5658078bfc5a923fb8d/drivers/gpu/drm/tinydrm/mipi-dbi.c#L235-L240

```c
        const u16 col_ofs = 2;
        const u16 row_ofs = 2;
	mipi_dbi_command(mipi, MIPI_DCS_SET_COLUMN_ADDRESS,
			 ((clip.x1 + col_ofs) >> 8) & 0xFF, (clip.x1 + col_ofs) & 0xFF,
			 ((clip.x2 + col_ofs) >> 8) & 0xFF, (clip.x2 + col_ofs - 1) & 0xFF);
	mipi_dbi_command(mipi, MIPI_DCS_SET_PAGE_ADDRESS,
			 ((clip.y1 + row_ofs) >> 8) & 0xFF, (clip.y1 + row_ofs) & 0xFF,
			 ((clip.y2 + row_ofs) >> 8) & 0xFF, (clip.y2 +row_ofs - 1) & 0xFF);
```

luRaichu, 2024-02-03:
> I think this is not the right place to make the change in our version of the kernel. I think it needs to be changed here: [...]

I just tested this now. No effect.

dlech, 2024-02-04:
How did you package and install the kernel? What does `uname -a` show?

luRaichu, 2024-02-05:
The kernel is properly installed.
```
Linux ev3dev 4.14.117-ev3dev-2.3.5-luraichu-fermion-tft-ev3+ #11 PREEMPT Mon Jan 15 17:14:30 UTC 2024 armv5tejl GNU/Linux
```

luRaichu, 2024-02-06:
Actually, this works! Turns out the release name had changed for the newly built kernel and dpkg refused to flash it. (oh, and `col_ofs` is supposed to be just 1, btw)
Here's my kernel build as a Debian package, with the screen offsets corrected for users of the Fermion 1.8" TFT.
[linux-image-4.14.117-ev3dev-2.3.5-luraichu-fermion-tft-ev3+_1_armel.deb.tar.gz](https://github.com/ev3dev/ev3dev/files/14186438/linux-image-4.14.117-ev3dev-2.3.5-luraichu-fermion-tft-ev3%2B_1_armel.deb.tar.gz)
Unarchive with `tar -xvf linux-image*.deb`, then install with `sudo dpkg -i linux-image*.deb`. If you're already running a custom kernel with an altered release name, you'll need to force install this.
[image]