ev3dev/ev3dev #1604: Streaming video with LEGO MovieMaker Camera

luRaichu, 2023-12-05:
Is it possible to stream video with the LEGO MovieMaker Camera, directly on the EV3?

dlech, 2023-12-06:
Low resolution streaming is possible.

Here is an example: https://github.com/G33kDude/ev3dev-web-remote

luRaichu, 2023-12-07:
I tried streaming with VLC:
`cvlc -vvv v4l2:///dev/video0 --noaudio --no-sout-audio --sout '#rtp{sdp=rtsp://:8080/}' :demux=h264`
This results with:
```
...
[b3b0d518] main demux meta debug: no meta reader modules matched
[004e3488] main input debug: Buffering 42%
[004e3488] main input debug: `v4l2:///dev/video0' successfully opened
[004e3488] main input debug: switching to sync mode
[004e3488] main input debug: Buffering 74%
[004e3488] main input debug: Stream buffering done (352 ms in 337 ms)
[b3b099c0] main decoder error: buffer deadlock prevented
[004e3488] main input debug: Decoder wait done in 6 ms
```
however when I go to VLC on my Mac and try to view `rtsp://ev3dev.local:8080` it doesn't work.
Just running `cvlc -vvv v4l2:///dev/video0` results in the camera feed displayed in Colour AsCii Art, though.

I also tried modifying the ev3dev-web-remote project, but I couldn't figure out how to update the feed without refreshing the page. fswebcam was too slow as well.

luRaichu, 2023-12-09:
I got RTMP streaming to work with ffmpeg!
`ffmpeg -i /dev/video0 -b:a 100 -pix_fmt yuv420p -tune zerolatency -s 178x146 -fflags nobuffer -f flv rtmp://192.168.2.1/live`
192.168.2.1 is the LAN address of the Mac which is running [an RTMP server](https://github.com/sallar/mac-local-rtmp-server). 
Then I view it on the Mac with `ffplay rtmp://127.0.0.1/live/`
Unfortunately there is a ~4 second latency using this method, even with the CPU running at 456MHz. Hopefully I can figure out how to cut down the bitrate? or switch to a more efficient protocol.

Nikolay-Zagrebin, 2024-06-04:
> Is it possible for the EV3 to stream video with the LEGO MovieMaker Camera to the LAN? https://lechnology.com/2016/02/lego-moviemaker-usb-camera-with-ev3-and-ev3dev/

I've been experimenting, and in principle, yes, everything is successful.
https://github.com/Nikolay-Zagrebin/LEGO-VISION-COMMAND-USB-Camera-with-EV3-and-ev3dev