ev3dev/ev3dev #1605: Brickrun using old version of micropython

CaptainFallaway, 2023-12-13:
**What are you trying to do?**

Run newest micropython version (2.0.0) with brickrun from vscode.

**What did you expect to happen**

I expected brickrun (which seems to quite outdated) to run the newest and upgraded version of micropython which i installed with apt earlier today.

The program should have...

Run and executed proberly

**What actually happened**

The program did not start since brickrun still says it's using micropython ver 1.11 

**Source Code**

Here is the program I am trying to run.

```
#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

from sys import exit

from time import sleep

# This program requires LEGO EV3 MicroPython v2.0 or higher.
# Click "Open user guide" on the EV3 extension tab for more information.

# Create your objects here.
ev3 = EV3Brick()
left = Motor(Port.B)
right = Motor(Port.C)
drive = DriveBase(left, right, 56, 114)

i = 0

# Write your program here.
while True:
    drive.curve(5, 360)

```

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS and version: Windows 21H2
 - Connection Type: USB atm
 - Software: Visual Studio Code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: See below

```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653602DD6
BOARD_INFO_TYPE=main
```

**Background Information**

I am trying to set up the robot to properly run the newest version of micropython so can run the latest pybrick library and of course use it's features and methods. I have connected the ev3 to my computer and used the ssh session to update it with the wired network using apt, I first fixed the sources list and then updated followed by upgrading. After that i did `sudo apt-get install pybricks-micopython3` which successfully installed ver 3.3.0. Then i rebooted since the upgrade nedded that and now i'm just trying to run it in vscode with no real idea of how to configure brickrun nor properly how to really execute stuff on this machine. I don't have much experience with the ev3 and i'm just getting started with it.

I have tried searching around with no real result, it just seemed to be going in a loop where i go to the same page with no avail. It also seems like this is an easy fix but i might have a problem with my patience or something.

dlech, 2023-12-13:
This version of Pybricks (v2.0) still reports the underlying Micropython version (v1.11) instead of the Pybricks version. So I think that, at least, is working as expected.

Pybricks v2.0 doesn't have the `DriveBase.curve()` method. And there are some issues with motor control in Pybricks v3 on ev3dev. See https://github.com/pybricks/support/issues/1035. So I'm afraid I don't have any good solutions. But I would suggest looking over the EV3-related discussions at https://github.com/orgs/pybricks/discussions and ask more questions there if you are still stuck.

CaptainFallaway, 2023-12-13:
Well thank you for the quick response, I will check out the discussion. Though I probably don't want to go ahead and use micropython now that I think about it and I also see that the EV3 can be programmed in so much more since it is an OS.

closed 2023-12-18 by CaptainFallaway