ev3dev/ev3dev #1606: ev3 doesn't from/with a SD card

drprofmedoekterofenfrische, 2024-01-03:
**What are you trying to do?**

I am trying to boot my ev3 from ev3dev.

**What did you expect to happen**

I expected the brick to boot from the SD as normal

**What actually happened**

The ev3 boots normal without the SD card, if I insert it, it instantly gets stuck at the MINDSTORMS booting screen, after I turn it off by removing the battery and the SD card it boots perfectly fine.

**Source Code**

I don't have a source code

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: Windows 10, Raspberry Pi OS
 - Connection Type: USB, Bluethoot
 - Software: Balena Etcher

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

**Background Information**

I am trying to install ev3dev because I want to build a robot that uses the PixyCam to catch red squares, I am unable to do this in ev3 Classroom or something like that. About 2-5 months when I last booted the brick with the same SD card it worked perfectly fine, now I am having the issue from above. I tried flashing it with Balena Etcher on Raspbarry Pi OS and Windows 10, both failed, Tried the same with Rufus, failed to.

dlech, 2024-01-03:
There is a known problem with recent versions of Etcher, but if other flashers aren't working either, then either you flashed the zip file as a raw file to the SD card instead of unzipping it first, you have a bad SD card or you have a bad SD card slot on the EV3.

gjhoppy, 2024-01-09:
What size is your SD card ?  I think they have to be 32Gb or smaller