ev3dev/ev3dev #1607: Connecting via Bluetootj

g-qwz, 2024-01-09:
**Describe the bug**
I'm trying to connect the ev3dev via Bluetooth following this tutorial:
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

The ev3dev disconnects after a few seconds of the connection. I can't find it on "All Network Connections" too.

It just appears on the Bluetooth Menu, but regarding this [Issue](https://github.com/ev3dev/ev3dev/issues/1519) we should not connect it from there.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Tutorial from Step 1-5
2. Do Step 6
3. Wait approx. 5seconds
4. Device Disconnects

**Desktop (please complete the following information):**
 - Linux Laptop with blueman and dnsmasq

**Robot (please complete the following information):**
 - EV3

dlech, 2024-01-10:
Step 6 is jut pairing, so it is expected to not stay connected after that step. Step 7 is where the network connection is made.

If the *Network Connection* button isn't show up then your laptop isn't properly indicating that network sharing is available. You may have to *Remove* the device, fix your laptop and then pair again.

g-qwz, 2024-01-10:
How can I be sure / fix that the Laptop is recognizing Network Share correctly? Are there e.g. hints to identify?

dlech, 2024-01-10:
It has been a while since I've done this, but I think you can use `bluetoothctl` to list the services provided by the adapter. The PAN (personal area network) service should be in that list. If it isn't there, I would try unconfiguring and reconfiguring things with blueman and make sure the dnsmasq package is actually installed.

g-qwz, 2024-01-10:
I tried that already and it didn't work; allthough after various reboots and waiting several hours it magically fixed itself.
I couldn't figure out the problem, but it somehow works now.