ev3dev/ev3dev #1608: build-essential is not available anymore

OtacilioNeto, 2024-01-21:
build-essential is not available. Following the step-by-step guide to install ev3dev-c throws this error message:
```
robot@ev3dev:~$ sudo apt-get install build-essential
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package build-essential is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'build-essential' has no installation candidate
robot@ev3dev:~$ 
```

this turns ev3dev useless for C coding.

dlech, 2024-01-21:
Since Debian stretch is no longer officially supported by Debian, the archive address changed. You will need to edit /etc/apt/sources.list on the EV3 to match

https://github.com/ev3dev/docker-library/blob/master/ev3dev-stretch/layers/debian/etc/apt/sources.list

Then run sudo apt update. Then installing should work. 

If you are compiling anything more than a simple program, I would recommend looking at using [docker](https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/) instead of compiling directly on the EV3 since it is quite slow and has limited memory.

OtacilioNeto, 2024-01-21:
Firstly, thank you very much. 
Is there any plan to update EV3DEV to a more recent version of Debian?

dlech, 2024-01-21:
> Is there any plan to update EV3DEV to a more recent version of Debian?

https://github.com/ev3dev/ev3dev/issues/1554