ev3dev/ev3dev #1610: Trying to get WiFi dongle 2357:0109 TP-Link TL WN823N RTL8192EU working

ajs54, 2024-02-12:
Hello.

I am trying to get a WiFi dongle Realtek 8192EU to work.  More precisely, `lsusb` returns 
```
Bus 001 Device 002: ID 2357:0109 TP-Link TL WN823N RTL8192EU
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
The command `ev3dev-sysinfo -m` returns:
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534B051A
BOARD_INFO_TYPE=main
```
I have `firmware-realtek` installed, as `apt list --installed | grep realtek` returns:
```
firmware-realtek/unknown,now 1:20161130-3+rpt3 all [installed]
```
I thus should have a WiFi module  RTl8192EU, according to http://packages.debian.org/bullseye/firmware-realtek. However, it is not started, and not even in the `/lib/modules/4.14.117-ev3dev-2.3.5-ev3/` sub-directory.

Should I try to install Mange module (https://github.com/Mange/rtl8192eu-linux-driver/archive/master.zip)?

ajs54, 2024-02-12:
Hello again.

In fact, I found `rtl8192eu_nic.bin` (which should be what I need - on need for Wakeup On Wifi) in `/lib/firmware/rtlwifi/`.

My question is thus: what should I do to get a kernel module from this file?

dlech, 2024-02-13:
It sounds like we added a driver that uses this firmware in #1115. The adapter was one with a different USB VID:PID though so not sure if will work. The driver is disabled by default on the EV3. To enable it, run `sudo nano /etc/modprobe.d/ev3.conf` to edit the config file and comment out the following line as shown. Save the file, then run `sudo modprobe rtl8xxxu` or reboot your EV3.

```
#blacklist rtl8xxxu
```

ajs54, 2024-02-14:
Hello, DLech, 

and thank you: I switched the blacklist in `/etc/modprobe.d/ev3.conf` (allowed `rtl8xxxu` instead of `rtl8192cu`), and started the module.  The WiFi is now activated, and a scan detected the available networks.

I still have to configure the connection, but I have a faculty engineer to help me with that.  I will confirm whether it works well or not, and close this thread.

ajs54, 2024-02-15:
Bad luck!

The engineer was supposed to help me once I was able to setup the WiFI, but he doesn't know connman!

I tried to set it up according to https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line, but when I ask to connect, I get `Error /net/connman/service/wifi_...: Invalid arguments`.  I tried to define a config file, and I then get `Method "Connect" with signature "" on interface "net.connman.Service" doesn't exist`, even if the agent is already on!

Could anybody help me?

dlech, 2024-02-15:
I would not try to manually create connman config files, they are hard to get right. 

I would check the logs to see if there are any more hints about what caused the initial error. Look for connman and wpa_supplicant.

```
sudo journalctl -xe
```

ajs54, 2024-02-16:
Hello.

Thank you for your help.

However, I tried at home with a simple protocol (WPA/2 PSK) and it works perfectly.  The faculty engineer gave me the informations about the WiFi dedicated to the Mindstorms, and it seems of the same type.

I still have to test it (in the beginning of next week).

If it works, I will close this thread (except if you want more informations about the PEAP v0 failure), and let you add this dongle as being tested (on https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles).

Have a nice week-end.

ajs54, 2024-02-21:
Hello!

Thank you very much:  you can add TP-Link TL WN823N RTL8192EU to the dongles working with Ev3dev.  I indeed have been able to configure both the WPA/2 PSK networks with WEP key (at home and university), and even eduroam - using https://wiki.archlinux.org/title/Network_configuration/Wireless#WPA2_Enterprise and https://wiki.archlinux.org/title/ConnMan#Connecting_to_eduroam_(802.1X).

About the dongle, it is a medium sized dongle (3,5 x 1,75 x 0,75 cm).

ajs54, 2024-02-21:
Thanks again.