ev3dev/ev3dev #1611: vscode ev3dev-browser extension better support

Natejoestev, 2024-02-18:
I am currently writing rust for the ev3dev system.
When i upload my project, it would be nice to specify a target to upload.
because it takes FOREVER to upload because of every, single, thing it uploads that aren't needed.

It would be nice to simply right-click a folder in the explorer, and click (upload to ev3dev)
it would be placed in `/home/robot/PROJECT NAME/FOLDER YOU SELECTED/...`

it also could just be a way to disable specific folders like `src` and `debug`, but that can be time consuming for each project.
maybe somehow it can sense it's a rust project and do something.

Natejoestev, 2024-02-18:
because 700/800 of the items are the debug stuff that isn't used

dlech, 2024-02-18:
There is a `ev3devBrowser.download.exclude` configuration option for this already.

Natejoestev, 2024-02-18:
oh. ill check it out

Natejoestev, 2024-02-18:
@dlech ok, that works well. it only downloads the binary.
but how would i add resources?
eg: this is my project directory:
```
PROJECT/
  src/
    main.rs
  resources/
    image.png
  target/
    arm....../
      release/
        BINARY

```
and i want this: on the ev3
```
/home/robot
  PROJECT/
    BINARY
    resources/
      *
```

dlech, 2024-02-18:
I would add a compile step on the computer to copy `resources` to `release/` so that it has the directory structure you want.

Natejoestev, 2024-02-18:
what i did what make a script to "build" this into a folder.
and i have `ev3dev-browser.include` set to `./build`

```ps1
cargo build --release
$_ = mkdir build -Force
copy .\target\armv5te-unknown-linux-musleabi\release\BINARY .\build -Force
$_ = mkdir build\resources -Force
copy .\resources\* .\build\resources -Force
```