ev3dev/ev3dev #1614: installing telnet deamo on ev3

eddieserko, 2024-02-19:
**Question**

How do you install telnet on the EV3

**Background Information**

I am trying to install Telnet for ussage of multiple programs using the protocol. 
I can not run sudo apt-get install telnetd:

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package telnetd

Maybe there is a newwer repository?

 - OS and version: [macOS 10.14]
 - Connection Type: [Netowrk]
 - Software: [terminal]

 - Device: [EV3]
 - Versions: [Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl GNU/Linux]

**System info (from `ev3dev-sysinfo`)**

Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=0016537F6743
BOARD_INFO_TYPE=main
```

Cheers
Edide

dlech, 2024-02-20:
The apt url needs to be updated. See instructions at https://github.com/ev3dev/ev3dev/issues/1608#issuecomment-1902726546.

luRaichu, 2024-02-24:
I don't know exacly what you're trying to do @eddieserko, but maybe you should look to more secure protocols like ssh for your project. Telnet is ancient and has security flaws.