ev3dev/ev3dev #1618: none of the command works

davimeglioli, 2024-03-04:
**What are you trying to do?**

install modules

**What did you expect to happen**

installing modules

**What actually happened**

robot@ev3dev:$ sudo apt-get install python-paramiko
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python-paramiko
robot@ev3dev:$ 

also the apt update command return this:

robot@ev3dev:$ sudo apt update
Err:1 http://archive.debian.org/debian stretch InRelease                
  Could not resolve 'archive.debian.org'
Err:2 http://archive.ev3dev.org/debian stretch InRelease                
  Could not resolve 'archive.ev3dev.org'
Reading package lists... Done        
Building dependency tree       
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://archive.debian.org/debian/dists/stretch/InRelease  Could not resolve 'archive.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
robot@ev3dev:$ 

i tried hundreds of solutions but none of them is working, i've already updated the etc/apt/sources.list file

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: Windows 11
 - Connection Type: USB
 - Software: Visual studio code

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 2020-04-10

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653497C0E
BOARD_INFO_TYPE=main
```

**Background Information**
<!-- Explaining the bigger picture helps give context to the problem -->

I am trying to ... because I want to build a robot that ...

davimeglioli, 2024-03-04:
yes, i see a 169.... ip, i don't have a usb ethernet adapter, how can i do with the usb from the pc?

davimeglioli, 2024-03-04:
my ip is 192.168.137...

robot@ev3dev:~$ hostname -I
192.168.137.45 
robot@ev3dev:~$ cat /etc/apt/sources.list
deb http://archive.debian.org/debian stretch main contrib non-free
#deb-src http://archive.debian.org/debian stretch main contrib non-free

deb http://archive.ev3dev.org/debian stretch main
#deb-src http://archive.ev3dev.org/debian stretch main

Growflavor, 2024-03-04:
> yes, i see a 169.... ip, i don't have a usb ethernet adapter, how can i do with the usb from the pc?

For me, when I have the 169.xxx I usually only need to go into the windows control panel network settings & toggle off the ICS for the EV3...then toggle it back on...the IP on the EV3 should switch to a 192.168.137.xxx after doing this...I don't have my EV3 plugged into my computer at the moment so it is not showing in the menu but here is where to look for it on your computer:

[image]

davimeglioli, 2024-03-04:
yes, i fixed it, now my ip  is 192.168.... but the commands don't work anyway

davimeglioli, 2024-03-04:
specifically I have to install the paramiko library and this is what it shows me

robot@ev3dev:~$ sudo apt-get install python-paramiko
[sudo] password for robot:
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python-paramiko

davimeglioli, 2024-03-04:
when i go in the connection settings on ev3dev i only see status : connected and not status : online, how can i fix it?

Growflavor, 2024-03-05:
Hmmm...the package you name just installed for me on an EV3 booted with a SDcard running "Linux ev3dev 4.14.117-ev3dev-2.3.5-ev3"  so at least I can confirm that as of today it is at least still possible...details further below...

try using apt rather than apt-get...

sudo apt update
sudo apt install python-paramiko

My recollection is that EV3 USB ICS worked ok even with Win 11, but if you have access to a Win10 computer definitely try USB ICS from Win 10. 

If that still does not work, then for ease of your projects, I very much suggest getting either a:

-- "USB2 WIFI to USB adapter": WIFI option is better if you want to dynamically control your bot via RPyC as it moves around & also have a connection out to the internet.  Though we also RPyC via Bluetooth connection...but at least with Windows Bluetooth IP connection seems to only be local with no way to connect to the internet via Bluetooth.

--or an "USB2 Ethernet to USB adapter" with ethernet cable to your hub/router

to plug directly into your EV3 USB port near the SDcard slot...since it makes life _so much easier_ than relying on the USB ICS 'headache' that you have encountered...

<details>
  <summary>Details & output from today for one of my EV3: robot@ev3dev:~$ sudo apt install python-paramiko</summary>
Today I used PuTTY (rather than VScode) to SSH to the EV3 from a Dell running Win 11  
& used an actual USB to ethernet adapter to connect the EV3 to the LAN & internet...

but in the past I have also succeeded using VScode ssh as well as the USB ICS at least from Win 10 to run sudo apt update & upgrade & install packages as long as I have a 192.168.137.xxx IP...
    
  ```javascript
  robot@ev3dev:~$ sudo apt install python-paramiko
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  python-cffi-backend python-cryptography python-enum34 python-idna python-ipaddress python-pyasn1
Suggested packages:
  python-cryptography-doc python-cryptography-vectors python-enum34-doc python-gssapi doc-base
The following NEW packages will be installed:
  python-cffi-backend python-cryptography python-enum34 python-idna python-ipaddress python-paramiko
  python-pyasn1
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 487 kB of archives.
After this operation, 3090 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.debian.org/debian stretch/main armel python-cffi-backend armel 1.9.1-2 [63.5 kB]
Get:2 http://archive.debian.org/debian stretch/main armel python-enum34 all 1.1.6-1 [35.0 kB]
Get:3 http://archive.debian.org/debian stretch/main armel python-idna all 2.2-1 [32.6 kB]
Get:4 http://archive.debian.org/debian stretch/main armel python-ipaddress all 1.0.17-1 [18.1 kB]
Get:5 http://archive.debian.org/debian stretch/main armel python-pyasn1 all 0.1.9-2 [51.8 kB]
Get:6 http://archive.debian.org/debian stretch/main armel python-cryptography armel 1.7.1-3+deb9u2 [175 kB]
Get:7 http://archive.debian.org/debian stretch/main armel python-paramiko all 2.0.0-1 [111 kB]
Fetched 487 kB in 3s (140 kB/s)
Selecting previously unselected package python-cffi-backend.
(Reading database ... 48481 files and directories currently installed.)
Preparing to unpack .../0-python-cffi-backend_1.9.1-2_armel.deb ...
Unpacking python-cffi-backend (1.9.1-2) ...
Selecting previously unselected package python-enum34.
Preparing to unpack .../1-python-enum34_1.1.6-1_all.deb ...
Unpacking python-enum34 (1.1.6-1) ...
Selecting previously unselected package python-idna.
Preparing to unpack .../2-python-idna_2.2-1_all.deb ...
Unpacking python-idna (2.2-1) ...
Selecting previously unselected package python-ipaddress.
Preparing to unpack .../3-python-ipaddress_1.0.17-1_all.deb ...
Unpacking python-ipaddress (1.0.17-1) ...
Selecting previously unselected package python-pyasn1.
Preparing to unpack .../4-python-pyasn1_0.1.9-2_all.deb ...
Unpacking python-pyasn1 (0.1.9-2) ...
Selecting previously unselected package python-cryptography.
Preparing to unpack .../5-python-cryptography_1.7.1-3+deb9u2_armel.deb ...
Unpacking python-cryptography (1.7.1-3+deb9u2) ...
Selecting previously unselected package python-paramiko.
Preparing to unpack .../6-python-paramiko_2.0.0-1_all.deb ...
Unpacking python-paramiko (2.0.0-1) ...
Setting up python-idna (2.2-1) ...
Setting up python-pyasn1 (0.1.9-2) ...
Setting up python-cffi-backend (1.9.1-2) ...
Setting up python-enum34 (1.1.6-1) ...
Setting up python-ipaddress (1.0.17-1) ...
Setting up python-cryptography (1.7.1-3+deb9u2) ...
Setting up python-paramiko (2.0.0-1) ...
robot@ev3dev:~$
  ```  
</details>

& your sources.list do look to correctly updated...

<details>
  <summary>here is the cat from my EV3:</summary>
    
  ```javascript

robot@ev3dev:~$ cat /etc/apt/sources.list
deb http://archive.debian.org/debian stretch main contrib non-free
#deb-src http://archive.debian.org/debian stretch main contrib non-free

deb http://archive.ev3dev.org/debian stretch main
#deb-src http://archive.ev3dev.org/debian stretch main
robot@ev3dev:~$
  ```  
</details>

<details>
  <summary>side note: My recollection is the main issue I have with Win 11 is when I try to have an EV3 talk with a WSL2 VM (e.g. We RPyC with the EV3 via Jupyter Notebook in a WSL2 Ubuntu)</summary>

something about the config change that came along with the new hypervisor firewall layer implemented in Win 11 strangely seems to block onboard localhost/ICS talk between the WSL2 VM & the ICS even though they are on the same machine...

but LAN addresses pass through fine...

where as with Win 10, 'default' settings seem to allow WSL2 to talk with both with ICS as well as LAN as well as 169. type IPs...

in any case, this is why I switched to using the ethernet to usb adapters since that method works with both Win 11 & Win 10)
  
</details>

Growflavor, 2024-03-05:
Also, might you be willing to update the title of this issue from "none of the command work" to something like "Installing packages via Windows USB Internet Connection Sharing"?  That might help other people with similar issues find this thread.

davimeglioli, 2024-03-05:
I managed to solve it. Windows sharing wasn't really working, I tried doing it with a Mac and it works

closed 2024-03-05 by davimeglioli