ev3dev/ev3dev #1619: Error installing with sudo apt

Alessandro-sys, 2024-03-12:
Hi everyone, I just installed ev3dev on my ev3 and I was trying to install pip

```
robot@ev3dev:~$ sudo apt update
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```

when I do sudo apt update no errors shows up, but when I try to install something it gives me this:

```
robot@ev3dev:~$ sudo apt install Python3-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package Python3-pip
```

I also noticed that my **sources.list** is empty, and I don't know what to put in. What should I do? 
thank you in advice for the help :)

dlech, 2024-03-12:
FYI, pip is broken since the Python version in ev3dev-stretch is too old it is no longer supported by PyPI. So even if you do manage to get the package installed, it won't work.

See https://github.com/ev3dev/ev3dev/issues/1608#issuecomment-1902726546 for how to fix the apt `sources.list` file.

Alessandro-sys, 2024-03-12:
So theres no way to install opencv there, right? 
Because I saw on YouTube that it is possible, but I have no idea how

dlech, 2024-03-12:
You can install the debian stretch version for Python 2

    sudo apt install python-opencv

IIRC, there isn't a Python 3 version available for Stretch.

Alessandro-sys, 2024-03-12:
it worked, thank you so much :)
what about numpy? is there any way to install it?

dlech, 2024-03-12:
For Python 3, you need `sudo apt install python3-numpy`.

Alessandro-sys, 2024-03-13:
I don't know what is wrong
I followed all the steps, installed all the required dependencies and tried to make a small program 
```
#!/usr/bin/env pybricks-micropython
import numpy as np
import cv2
```

and I still get 
```
Starting: brickrun --directory="/home/robot/prova" "/home/robot/prova/main.py"
----------
Traceback (most recent call last):
  File "/home/robot/prova/main.py", line 2, in <module>
ImportError: no module named 'numpy'
----------
Exited with error code 1.
```
:(

closed 2024-03-15 by Alessandro-sys